SCADA-LTS/Scada-LTS #1725: Advertising is much more important than coding.....

jasonlee9900, 2021-06-03:
There are very little introduce for Scada-LTS, what's new function developed than old old mango M2M?
Could you make some screenshot to introduce what's new features and advantages in Scada-LTD compare with old Mango M2M?
An good advertising example like https://github.com/thingsboard/thingsboard can attract more people to join in Scada-LTD and support you. 
Anyway, the installation of Scada-LTS is very complex and difficult compare to old mango M2M or scadaBR1.0. The reason for popular of ScadaBR 1.0 is because it's one click installation so simple and easy, hope LTS will develop the same function.

Limraj, 2023-06-03:
Hi @jasonlee9900,
Please visit and subscribe to our youtube channel:
https://www.youtube.com/@ScadaLTS

This is how the changes are written in the issue titles, only our improvements are listed there, so after that you can see more or less what has been improved and developed:
https://github.com/SCADA-LTS/Scada-LTS/releases

If you have a specific question or a problem that made it difficult to use ScadaBR, you can ask, I will try to answer if Scada-LTS has been improved in this aspect.

Regards,
Kamil Jarmusik

closed 2023-06-12 by Limraj

fabiodurao, 2023-06-13:
Hello, to help other users I will also make an overview of the various new features that exist in Scada-LTS, but keep in mind that there are several others that are in the background, or to improve stability that I will not be able to mention, and only by looking at the changelogs even to understand.

It is also worth mentioning that Mango M2M and ScadaBR had a parallel development until version 1.0, after that Mango became proprietary and had a lot of development in another direction, ScadaBR had a small upgrade to version 1.1 and from there the Scada-LTS began to be developed.

You are right when you say that Scada-LTS is complex to install, there is only one correction missing, it WAS complex. Now there is an installer for windows (https://github.com/SCADA-LTS/windows-installer/releases/tag/v1.0.5) of the type next, next and finish that is even better than the ScadaBR installer because it already comes native with the Mysql 8 database. An installer for linux was also recently created (https://github.com/SCADA-LTS/linux-installer). There are still complex options like manual or docker installation for advanced users, but it is no longer necessary to resort to these methods to access and test the software or apply it to a project.

Improvements/News:
- Update to tomcat 9, java 11;
- Installer for windows already with mysql 8 native - no longer uses derby;
- Improvement of stability in general and in the acquisition of values;
- Improvement in the functioning of the graphical representations, the elements update individually, you don't need to update the entire page with each update, as it was before;
- User profile, was added in ScadaBR 1.1 and improved in Scada-LTS;
- Everywhere you can choose a datapoint, there is a search field that has already been bringing the datapoints as you type the name;
- Modbus Serial working, there was a problem in scadabr 1.1 but it was fixed;
- Bug fix in OPC DA, apparently inherited from mango;
- Addition of the MQTT protocol for integration with IoT;
- Integration with the AmCharts graphic library that enables better and more beautiful graphics and different areas of the system (component in the graphical representation, watch list, among others);
- New components for graphical representations (normal graph, graph with interval definition, isAlive - to show the client if the server is alive or dead, auto/manual);
- Graphic components with x and y coordinate function and with layer/depth control with the z variable;
- Options for defining the resolution of the graphic representation before uploading the background image (eg: 1920x1080, 640x480 ...);
- New interface for writing css stylesheet to facilitate customization of graphic elements;
- Options per user and general to open graphical representation in fullscreen and hide the text "press ctrl+shif+f to exit fullscreen";
- New graphical representation based on component mapping within an SVG image, still under development, but with great potential for improvement - the traditional representation still exists (watch to understand https://youtu.be/1za1GMlaUJk);
- Integration with sms gateway and sms-type event handler;
- Greater user control over the theme and language used when logging in;
- Improvements in languages and admin control;
- Possibility to add graphics library without having to stop tomcat;
- Addition of confirmation step to delete screens, datasources, datapoints among many other elements that were formerly deleted in accidental clicks;
- New layout of the entire system being built with Vue.js, still under development;
- New options within the http publisher for integration with API's (json, authentication);
- Improvements in http receiver (authentication, sleep etc);
- Improvements in the sql datasource. It is now possible to connect to a DB using JNDI, which allows controlling the number of connections among other parameters. The same feature is also natively used for Scada-LTS to communicate with its default database;
- Improvements and bug fixes in the json and .zip export and import tool;
- Support for bug fixes and continuous improvements on github, which allows users to communicate with developers directly;

I believe that already gives you a north to understand where the system is heading, I hope I helped.