SCADA-LTS/Scada-LTS #1734: Vulnerabilities from ScadaBR

celsou, 2021-06-11:
Hello. I'm from the ScadaBR Forum. Recently a security researcher reported two ScadaBR vulnerabilities on our forum. I can't say if these vulnerabilities also affect Scada-LTS, but I thought it was important to warn.

The forum post about the problem is here (in Portuguese): http://forum.scadabr.com.br/t/report-falhas-de-seguranca-em-versoes-do-scadabr/3615 

I implemented fixes for these vulnerabilities in a ScadaBR fork that I develop. Here are the commits, which might help you see if the issue also affects Scada-LTS: 
https://github.com/celsou/ScadaBR-EF/commit/9e22a1aae3c9aa538549ba4aeada524ff10427d2
https://github.com/celsou/ScadaBR-EF/commit/43906b81173c0cf38a6a34050baa1ec6087ad7e6

celsou, 2021-06-14:
I have found that to fix the background image vulnerability it is necessary to also limit the allowed file extensions. In this commit I implemented a whitelist of file extensions:
https://github.com/celsou/ScadaBR-EF/commit/4438cd066a4baad6ea9facbd608d31f71e833ee1

SoftQ, 2021-06-14:
Hello

Limraj, 2021-06-16:
#1384

AdisonCavani, 2022-02-24:
Image validation was fixed in #2089

commit 33f5ec7c (2022-02-25): Merge pull request #2103 from SCADA-LTS/fix/#1734_XSS_Vulnerabilities_from_ScadaBR_alt

Fix #1734 - XSS Vulnerabilities from ScadaBR
closed 2022-02-24 by AdisonCavani