SCADA-LTS/Scada-LTS #1738: TLS Update to Version 1.3

fabiodurao, 2021-06-14:
The current version of TLS is no longer supported by some systems, so it is important to upgrade to the current version.

Example: Integration with Telegram API through http Publisher previously worked, now Telegram only accepts if TLS is version 1.2 or higher, which makes this integration impossible through traditional ways.

Problems are also starting to occur in integrations with email servers for sending notifications and in integrations with apps. I used Blynk app as a mobile interface for applications with ScadaLTS, which is no longer possible.

WagnerDeQueiroz, 2021-06-14:
Yes. I installed scadaLTS this weekend and I found the same problem. We use telegram messages to alert users of some problems and we cannot use company email to send reports because customer email account needs TLS to work.

SoftQ, 2021-06-15:
Hello
@fabiodurao - it may be rather a matter of Java version. Please write which version of Java and Tomcat  You are experiiencing the problem:

Java Version | SSL/TLS Default | Other Supported Versions
-- | -- | --
Java 6 | TLS 1.0 | TLS 1.1 (update 111 and later), SSLv3.0*
Java 7 | TLS 1.0 | TLS 1.2, TLS 1.1, SSLv3.0*
Java 8 | TLS 1.2 | TLS 1.1, TLS 1.0, SSLv3.0*

@WagnerDeQueiroz, @fabiodurao  - please run https://github.com/SCADA-LTS/Scada-LTS/releases/tag/v2.6.4 version on newest Java 11 and Tomcat 9, and confirm whether the problem persists.

Limraj, 2021-06-15:
https://mail.openjdk.java.net/pipermail/jdk8u-dev/2020-October/012817.html
openjdk 8u272 supported TLSv1.3

WagnerDeQueiroz, 2021-06-15:
I tested the last release 2.6.4 with last stable OpenJDK8 on Ubuntu machine without sucess...

my java setup is OpenJDK 1.8.0_292-b10
Tomcat7.0.81
Ubuntu 20.04 Desktop. 

I followed the tutorial https://github.com/SCADA-LTS/Scada-LTS/wiki/Install-Scada-LTS-on-linux,but not used apt because into ubuntu 20, don't have packages from ubuntu repositories.

I used the same url to test the telegram message on browser and by Browser works, but at ScadaLTS http sender, we got the error:

ERROR: Invalid response code: 400

SoftQ, 2021-06-16:
Please try to upgrade to the newest Tomcat 7 and let know if it works.  In case not, please provide the URL/instruction how to reproduce the issue.

Patrykb0802, 2023-07-22:
Hello
@fabiodurao, do you know if this problem still occurs ? And if you don't know, can you explain how this integration works ?
Best regards,
Patryk

closed 2024-05-24 by Patrykb0802