SCADA-LTS/Scada-LTS #1752: Docker do not work, so as to Linux installation

jasonlee9900, 2021-06-18:
Docker do not work, see picture:
[image]

And I spent 7 days to try install it manually on Ubuntu 20.04 according to your instruction, it do not work again. 
may you write a shell to install it automatically?

radek2s, 2021-06-18:
Hello @jasonlee9900 could you share with us more details about your docker installation? Which image do you tried to run? 

About running on Ubuntu 20.04 there is a problem with MySQL server installation. So for on host deployment on Ubuntu 20.04 we suggest to use MySQL server docker image 
```docker pull mysql/mysql-server:5.7```

davidlane-iteratec, 2021-06-18:
I am having the same problem. I followed the instructions here: https://github.com/SCADA-LTS/Scada-LTS/wiki/Run-ScadaLTS-on-docker---instruction
I am getting the same log as above. The war fails to start. 
The instructions say: docker pull scadalts/scadalts:latest and run it

I looked into docker hub and latest is 2 Months old. "release" is however only 3 days old. When I try it with
docker run -p 8199:8080 scadalts/scadalts:release
then it works.

Limraj, 2021-06-18:
More details about this issue are provided in: #1687

jasonlee9900, 2021-06-18:
hi, all: 
I got the problem solved according to davidlane-iteratec's comment, thanks. 
Seems a lot of works have been done for this system. and it's huge workload for rebuild the SCADA,  look forward to see your new updates!

radek2s, 2021-08-20:
We had a problem due to build environment migration. We moved from Travis to GitHub Actions and some features like building docker images were not included in this new workflow. Now we have finished works related to that topic and there is a new way how to use our docker images. 

We moved to `docker-compose` solution that is described on our [Wiki](https://github.com/SCADA-LTS/Scada-LTS/wiki/Scada-LTS-docker-compose-tutorial). But we are waiting for merge to 'master' branch to make that stable. But if you want to check if it is working you can find the solution [here](https://github.com/SCADA-LTS/Scada-LTS/issues/1799). Just download a environment configuration [docker-compose.yml](https://github.com/SCADA-LTS/Scada-LTS/blob/release/2.6.9/docker-compose.yml) file. Change tag of 'scadalts' from `master` to `pr-1849` and start a 'database' container. 
```
docker-compose up database
```
Then you should start the 'scadalts' container and the application should be accessible on `localhost:8080\Scada-LTS`. 
```
docker-compose up scadalts
```

closed 2022-11-01 by Limraj