SCADA-LTS/Scada-LTS #1761: Improve AmChart performance and User Experience

radek2s, 2021-06-21:
**Expected Behavior**
- Display last Data Point value (even if in specific period of time there was no update)
- Display notification that will inform user that specific data point does not contain any values.
- Inform about "not reliable data" instead of display nothing.

**Actual Behavior**
Charts are loading very long for huge amount of data. Additionally if there are any data in specific DataPoint there is no message about that and user may think that something does not work.

*Specifications*
Version: 2.6.7
Platform: Linux

fabiodurao, 2021-06-21:
This is very interesting, today when we want a chart that shows data from datapoints that don't change much, with binary and numeric setpoints, we choose the option to register "all data". This causes storage to increase unnecessarily.

Last time I talked about this with the developers, I was told that the intention was not to show data that was not actually read, giving false impression of update, but this solution proposed above seems to solve all related issues.

radek2s, 2021-07-06:
Hope that this commits will also reduce the CPU and memory usage issue #1478 reported by @OlleRatu.
Resolve #1478

closed 2021-07-06 by radek2s