SCADA-LTS/Scada-LTS #1795: Are there any plans for a Windows installer?

celsou, 2021-07-13:
Hello Scada-LTS team

I would like to know if you have any plans to create a Windows installer for future versions of Scada-LTS. At least here in Brazil many people who use ScadaBR are starting in the automation area and start with simple projects involving Arduino and Windows. An installer for Windows would be very useful to help these and others to use Scada-LTS.

If you need a base to create an installer for Windows, you can use the installer I created for my version of ScadaBR, [Innacio](https://github.com/celsou/innacio). It is a simple installer made with Inno Setup, but it can serve as a starting point for more complete installers.

Finally, I want to congratulate all the Devs for the amazing work you are doing on Scada-LTS. Greetings from Brazil.

farmsid, 2021-07-13:
Sim sim, já to de olho.

Limraj, 2021-07-14:
Thank you for appreciating our work, we are very pleased to hear that! 
The problem is that we have limited resources and we must carefully choose what to do. If you know how to prepare such an installer, you can write to us here or propose a change in the form of a pullrequest. We'll be grateful for help. 
Greetings.

celsou, 2021-07-15:
I don't have time for now and I can't promise anything, but when I have more time I will try to adapt "Innacio" to Scada-LTS. Greetings.

Limraj, 2023-06-03:
Hi @celsou,
Here we have prepared an installer for windows:
https://github.com/SCADA-LTS/windows-installer 

Regards,
Kamil Jarmusik

closed 2023-06-03 by Limraj