SCADA-LTS/Scada-LTS #1799: Problem when trying to run with Docker

renatocan, 2021-07-14:
Problem when trying to run with Docker

I've followed the steps described at https://github.com/SCADA-LTS/Scada-LTS/wiki/Run-ScadaLTS-on-docker---instruction but got some errors when creating the docker container.

I think the main error is related to a migration failure:
```
Migration of schema `scadalts` to version 1.1 - ViewsHierarchy failed! Please restore backups and roll back database and code! 
```

Am I missing something here? I assume the docker instalation should be straighforward.

Here's the outputs of the docker commands:
```
➜  ~ sudo docker pull scadalts/scadalts:release            
release: Pulling from scadalts/scadalts
Digest: sha256:d86f9c4a3313af76bd0d357b495f1b8a94d326d166b9bfbe485467ea277953e3
Status: Image is up to date for scadalts/scadalts:release
docker.io/scadalts/scadalts:release
➜  ~ sudo docker run -p 81:8080 scadalts/scadalts:release
chown: cannot access '/var/run/mysqld': No such file or directory
 * Starting MySQL database server mysqld
   ...done.
mysql: [Warning] Using a password on the command line interface can be insecure.
Tomcat started.
Tomcat started.
tail: cannot open '100' for reading: No such file or directory
==> /opt/scadalts/logs/catalina.out <==
Jul 14, 2021 9:31:04 PM org.apache.tomcat.jdbc.pool.interceptor.SlowQueryReport reportFailedQuery
WARNING: Failed Query Report SQL=select count(*) from users; time=13 ms;
INFO  2021-07-14T21:31:04,764 (com.serotonin.mango.db.MySQLAccess.newDatabaseCheck:107) - schemaExist:false 
Jul 14, 2021 9:31:04 PM org.apache.tomcat.jdbc.pool.interceptor.SlowQueryReport reportFailedQuery
WARNING: Failed Query Report SQL=select count(*) from schema_version; time=1 ms;
INFO  2021-07-14T21:31:04,766 (com.serotonin.mango.db.MySQLAccess.newDatabaseCheck:115) - BaseLineNotExist:true 
WARN  2021-07-14T21:31:04,776 (org.flywaydb.core.internal.logging.slf4j.Slf4jLog.warn:53) - Use of dots (.) as path separators will be deprecated in Flyway 7. Path: org.scada_lts.dao.migration.mysql 
Jul 14, 2021 9:31:32 PM org.apache.tomcat.jdbc.pool.interceptor.SlowQueryReport reportSlowQuery
WARNING: Slow Query Report SQL=alter table userEvents add constraint userEventsFk2 foreign key (userId) references users(id) on delete cascade;; time=1843 ms;
NOTE: Picked up JDK_JAVA_OPTIONS:  --add-opens=java.base/java.lang=ALL-UNNAMED --add-opens=java.rmi/sun.rmi.transport=ALL-UNNAMED
Jul 14, 2021 9:31:56 PM org.apache.catalina.startup.VersionLoggerListener log
INFO: Server version:        Apache Tomcat/7.0.85
Jul 14, 2021 9:31:56 PM org.apache.catalina.startup.VersionLoggerListener log
INFO: Server built:          Feb 7 2018 18:52:33 UTC
Jul 14, 2021 9:31:56 PM org.apache.catalina.startup.VersionLoggerListener log
INFO: Server number:         7.0.85.0
Jul 14, 2021 9:31:56 PM org.apache.catalina.startup.VersionLoggerListener log
INFO: OS Name:               Linux
Jul 14, 2021 9:31:56 PM org.apache.catalina.startup.VersionLoggerListener log
INFO: OS Version:            5.12.15-arch1-1
Jul 14, 2021 9:31:56 PM org.apache.catalina.startup.VersionLoggerListener log
INFO: Architecture:          amd64
Jul 14, 2021 9:31:56 PM org.apache.catalina.startup.VersionLoggerListener log
INFO: Java Home:             /usr/lib/jvm/java-11-openjdk-amd64
Jul 14, 2021 9:31:56 PM org.apache.catalina.startup.VersionLoggerListener log
INFO: JVM Version:           11.0.11+9-Ubuntu-0ubuntu2.18.04
Jul 14, 2021 9:31:56 PM org.apache.catalina.startup.VersionLoggerListener log
INFO: JVM Vendor:            Ubuntu
Jul 14, 2021 9:31:56 PM org.apache.catalina.startup.VersionLoggerListener log
INFO: CATALINA_BASE:         /opt/scadalts
Jul 14, 2021 9:31:56 PM org.apache.catalina.startup.VersionLoggerListener log
INFO: CATALINA_HOME:         /opt/scadalts
Jul 14, 2021 9:31:56 PM org.apache.catalina.startup.VersionLoggerListener log
INFO: Command line argument: --add-opens=java.base/java.lang=ALL-UNNAMED
Jul 14, 2021 9:31:56 PM org.apache.catalina.startup.VersionLoggerListener log
INFO: Command line argument: --add-opens=java.rmi/sun.rmi.transport=ALL-UNNAMED
Jul 14, 2021 9:31:56 PM org.apache.catalina.startup.VersionLoggerListener log
INFO: Command line argument: -Djava.util.logging.config.file=/opt/scadalts/conf/logging.properties
Jul 14, 2021 9:31:56 PM org.apache.catalina.startup.VersionLoggerListener log
INFO: Command line argument: -Djava.util.logging.manager=org.apache.juli.ClassLoaderLogManager
Jul 14, 2021 9:31:56 PM org.apache.catalina.startup.VersionLoggerListener log
INFO: Command line argument: -Djdk.tls.ephemeralDHKeySize=2048
Jul 14, 2021 9:31:56 PM org.apache.catalina.startup.VersionLoggerListener log
[... 95 lines cut ...]
Jul 14, 2021 9:32:02 PM org.apache.catalina.startup.HostConfig deployDirectory
INFO: Deploying web application directory /opt/scadalts/webapps/host-manager
Jul 14, 2021 9:32:02 PM org.apache.catalina.startup.TldConfig execute
INFO: At least one JAR was scanned for TLDs yet contained no TLDs. Enable debug logging for this logger for a complete list of JARs that were scanned but no TLDs were found in them. Skipping unneeded JARs during scanning can improve startup time and JSP compilation time.
Jul 14, 2021 9:32:02 PM org.apache.catalina.startup.HostConfig deployDirectory
INFO: Deployment of web application directory /opt/scadalts/webapps/host-manager has finished in 105 ms
Jul 14, 2021 9:32:02 PM org.apache.catalina.startup.HostConfig deployDirectory
INFO: Deploying web application directory /opt/scadalts/webapps/docs
Jul 14, 2021 9:32:02 PM org.apache.catalina.startup.TldConfig execute
INFO: At least one JAR was scanned for TLDs yet contained no TLDs. Enable debug logging for this logger for a complete list of JARs that were scanned but no TLDs were found in them. Skipping unneeded JARs during scanning can improve startup time and JSP compilation time.
Jul 14, 2021 9:32:02 PM org.apache.catalina.startup.HostConfig deployDirectory
INFO: Deployment of web application directory /opt/scadalts/webapps/docs has finished in 89 ms
Jul 14, 2021 9:32:02 PM org.apache.coyote.AbstractProtocol start
INFO: Starting ProtocolHandler ["http-bio-8080"]
Jul 14, 2021 9:32:02 PM org.apache.coyote.AbstractProtocol start
INFO: Starting ProtocolHandler ["ajp-bio-8009"]
Jul 14, 2021 9:32:02 PM org.apache.catalina.startup.Catalina start
INFO: Server startup in 6267 ms
```

I've also tried to run using docker-compose with the yaml file at https://github.com/SCADA-LTS/Scada-LTS but also got problems. It says the file Scada-LTS.war cannot be found.

Thanks,

Renato.

Limraj, 2021-07-15:
Hello, try:
```
docker images
docker rmi -f scadalts_release_image_id 
docker run -it -p 81:8080 scadalts/scadalts:release /root/start.sh
```
If application is started, open: http://localhost:81/ScadaBR/
This is how I started it, it looks okay.

renatocan, 2021-07-20:
Thank you for your suggestion. I've tried to run this way, but got the same error message:

```
Unable to find image 'scadalts/scadalts:release' locally
release: Pulling from scadalts/scadalts
4bbfd2c87b75: Pull complete 
d2e110be24e1: Pull complete 
889a7173dcfe: Pull complete 
c7b2dd57c0d1: Pull complete 
434cf566edbb: Pull complete 
ec3bdd6c3ce2: Pull complete 
34616b3fc948: Pull complete 
5fdb7ae32202: Pull complete 
2893ee32081a: Pull complete 
ea5bd0b5b399: Pull complete 
462f9cfc5265: Pull complete 
ec9d3108cb7c: Pull complete 
3ddc213b6010: Pull complete 
Digest: sha256:d86f9c4a3313af76bd0d357b495f1b8a94d326d166b9bfbe485467ea277953e3
Status: Downloaded newer image for scadalts/scadalts:release
chown: cannot access '/var/run/mysqld': No such file or directory
 * Starting MySQL database server mysqld                                        No directory, logging in with HOME=/
                                                                         [ OK ]
mysql: [Warning] Using a password on the command line interface can be insecure.
Using CATALINA_BASE:   /opt/scadalts
Using CATALINA_HOME:   /opt/scadalts
Using CATALINA_TMPDIR: /opt/scadalts/temp
Using JRE_HOME:        /usr
Using CLASSPATH:       /opt/scadalts/bin/bootstrap.jar:/opt/scadalts/bin/tomcat-juli.jar
Tomcat started.
Using CATALINA_BASE:   /opt/scadalts
Using CATALINA_HOME:   /opt/scadalts
Using CATALINA_TMPDIR: /opt/scadalts/temp
Using JRE_HOME:        /usr
Using CLASSPATH:       /opt/scadalts/bin/bootstrap.jar:/opt/scadalts/bin/tomcat-juli.jar
Tomcat started.
tail: cannot open '100' for reading: No such file or directory
==> /opt/scadalts/logs/catalina.out <==
WARNING: Failed Query Report SQL=select count(*) from schema_version; time=1 ms;
INFO  2021-07-20T21:49:24,669 (com.serotonin.mango.db.MySQLAccess.newDatabaseCheck:115) - BaseLineNotExist:true 
WARN  2021-07-20T21:49:24,679 (org.flywaydb.core.internal.logging.slf4j.Slf4jLog.warn:53) - Use of dots (.) as path separators will be deprecated in Flyway 7. Path: org.scada_lts.dao.migration.mysql 
Jul 20, 2021 9:49:42 PM org.apache.tomcat.jdbc.pool.interceptor.SlowQueryReport reportSlowQuery
WARNING: Slow Query Report SQL=alter table mangoViewUsers add constraint mangoViewUsersFk2 foreign key (userId) references users(id) on delete cascade;; time=1527 ms;
Jul 20, 2021 9:49:57 PM org.apache.tomcat.jdbc.pool.interceptor.SlowQueryReport reportSlowQuery
WARNING: Slow Query Report SQL=alter table events add constraint eventsFk1 foreign key (ackUserId) references users(id);; time=3888 ms;
Jul 20, 2021 9:49:58 PM org.apache.tomcat.jdbc.pool.interceptor.SlowQueryReport reportSlowQuery
WARNING: Slow Query Report SQL=create table userEvents (eventId int not null,userId int not null,silenced char(1) not null,primary key (eventId, userId)) ENGINE=InnoDB;; time=1869 ms;
NOTE: Picked up JDK_JAVA_OPTIONS:  --add-opens=java.base/java.lang=ALL-UNNAMED --add-opens=java.rmi/sun.rmi.transport=ALL-UNNAMED
Jul 20, 2021 9:50:20 PM org.apache.catalina.startup.VersionLoggerListener log
INFO: Server version:        Apache Tomcat/7.0.85
Jul 20, 2021 9:50:20 PM org.apache.catalina.startup.VersionLoggerListener log
INFO: Server built:          Feb 7 2018 18:52:33 UTC
Jul 20, 2021 9:50:20 PM org.apache.catalina.startup.VersionLoggerListener log
INFO: Server number:         7.0.85.0
Jul 20, 2021 9:50:20 PM org.apache.catalina.startup.VersionLoggerListener log
INFO: OS Name:               Linux
Jul 20, 2021 9:50:20 PM org.apache.catalina.startup.VersionLoggerListener log
INFO: OS Version:            5.12.15-arch1-1
Jul 20, 2021 9:50:20 PM org.apache.catalina.startup.VersionLoggerListener log
INFO: Architecture:          amd64
Jul 20, 2021 9:50:20 PM org.apache.catalina.startup.VersionLoggerListener log
INFO: Java Home:             /usr/lib/jvm/java-11-openjdk-amd64
Jul 20, 2021 9:50:20 PM org.apache.catalina.startup.VersionLoggerListener log
INFO: JVM Version:           11.0.11+9-Ubuntu-0ubuntu2.18.04
Jul 20, 2021 9:50:20 PM org.apache.catalina.startup.VersionLoggerListener log
INFO: JVM Vendor:            Ubuntu
Jul 20, 2021 9:50:20 PM org.apache.catalina.startup.VersionLoggerListener log
INFO: CATALINA_BASE:         /opt/scadalts
Jul 20, 2021 9:50:20 PM org.apache.catalina.startup.VersionLoggerListener log
INFO: CATALINA_HOME:         /opt/scadalts
[... 96 lines cut ...]
Jul 20, 2021 9:50:23 PM org.apache.catalina.startup.HostConfig deployDirectory
INFO: Deploying web application directory /opt/scadalts/webapps/examples
Jul 20, 2021 9:50:24 PM org.apache.catalina.startup.TldConfig execute
INFO: At least one JAR was scanned for TLDs yet contained no TLDs. Enable debug logging for this logger for a complete list of JARs that were scanned but no TLDs were found in them. Skipping unneeded JARs during scanning can improve startup time and JSP compilation time.
Jul 20, 2021 9:50:24 PM org.apache.catalina.startup.HostConfig deployDirectory
INFO: Deployment of web application directory /opt/scadalts/webapps/examples has finished in 198 ms
Jul 20, 2021 9:50:24 PM org.apache.catalina.startup.HostConfig deployDirectory
INFO: Deploying web application directory /opt/scadalts/webapps/host-manager
Jul 20, 2021 9:50:24 PM org.apache.catalina.startup.TldConfig execute
INFO: At least one JAR was scanned for TLDs yet contained no TLDs. Enable debug logging for this logger for a complete list of JARs that were scanned but no TLDs were found in them. Skipping unneeded JARs during scanning can improve startup time and JSP compilat [... cut ...]
Jul 20, 2021 9:50:24 PM org.apache.catalina.startup.HostConfig deployDirectory
INFO: Deployment of web application directory /opt/scadalts/webapps/docs has finished in 71 ms
Jul 20, 2021 9:50:24 PM org.apache.coyote.AbstractProtocol start
INFO: Starting ProtocolHandler ["http-bio-8080"]
Jul 20, 2021 9:50:24 PM org.apache.coyote.AbstractProtocol start
INFO: Starting ProtocolHandler ["ajp-bio-8009"]
Jul 20, 2021 9:50:24 PM org.apache.catalina.startup.Catalina start
INFO: Server startup in 4033 ms
```

radek2s, 2021-08-20:
Hello @renatocan 

Based on your Issue we provided some improvements in our building process. We rebuild our docker container from one image with Ubuntu system to smaller based on the Tomcat image. Due to that we need to have an external DataBase image that will be connecting to Scada-LTS. To make that simple we provided a `docker-compose` solution. 

You can download the docker environment configuration from [our repository](https://github.com/SCADA-LTS/Scada-LTS/blob/release/2.6.9/docker-compose.yml). But if you want to check if everything is working you have to change the `scadalts/scadalts` image tag from 'master' to 'pr-1849' that is the latest version of our code. After that just type:
```
docker-compose up -d database
```
And then when the MySQL server is running start the our Scada application using:
```
docker-compose up scadalts
```

Now everything should be working. If you could share your updates if that solution has helped you.

Regards

renatocan, 2021-10-19:
Hi @radek2s, 

Your suggestion solved the issue. In fact, I've seen that the `:latest` scadalts image is also working. As I took a lot of time to test it, I think the changes have been committed to it.

Thank you very much and sorry for the late reply.

Best,

Renato.