SCADA-LTS/Scada-LTS #1822: Release 2.6.8 manual install on RPi: tomcat & java versions?

KMKAR, 2021-07-25:
Hi all. I'm going to start from scratch a new (manual installation) over a RPi (3b+).
I just want to confirm that the versions for both tomcat and java described on [https://github.com/SCADA-LTS/Scada-LTS/wiki/Install-Scada-LTS-on-linux](url) are still valid with the newest releases of ScadaLTS.

Thanks to everyone, regards.

radek2s, 2021-07-26:
Hello @KMKAR 

The [installation tutorial](https://github.com/SCADA-LTS/Scada-LTS/wiki/Install-Scada-LTS-on-linux) is up-to-date and should be working for Java 8 and Tomcat 7. But there is also an option to run Scada-LTS with Java 11 and Tomcat 8, but that configuration has not been described on our wiki. 

PS.
We are not sure how it will work on RPi so we do not guarantee that the dependencies are compatible with ARM architecture. 

Regards,
Radek

carl1961, 2021-07-26:
I been trying to get it installed on RPI4 buster
1. Install Tomcat server
apt-get install tomcat7 ( ver 7 no longer supported, won't install ) you have to use apt-get install tomcat8
2. Install MySQL server     
apt-get install mysql-server    MySQL not supported either on buster , you have to use  sudo apt install mariadb-server but default database does not install on  /opt  you have to edit   nano  /etc/mysql/mysql.conf.d/50-server.cnf  and change it in there from the default  /var/lib/mysql  /opt

carl1961, 2021-07-27:
seen this   ScadaLTS Ubuntu18 04

https://www.youtube.com/watch?v=acYo8OCG0Vo
https://drive.google.com/file/d/1XRd-khQX3Ts4oIBxNkkMVf4_8rp0o8eC/view

carl1961, 2021-07-29:
@ Tip
Collaborator
radek2s  This guy hsainnos has the ScadaLTS running on a Open Source Project   Looks awesome too. very detailed https://github.com/hsainnos/LICSTER

carl1961, 2021-08-02:
ScadaLTS Installer for Rasberry PI      https://github.com/carl1961/ScadaLTS_RPI_Installer

KMKAR, 2021-08-06:
@carl1961 that's a very cool script to automate the install process. I'll give it a try. So far, on my end, I've a RPi 3 b+, latest RaspiOS, openjdk 11 and tomcat 8.5.38
I have no issues, just a little load average higher than expected but I think that in an rpi 4 this shouldn't be a problem. I'm just using 3 b+ in a lab, for a proof-of-concept.
I might fork/contribute to your repo, again, great work!

closed 2022-03-22 by Limraj