SCADA-LTS/Scada-LTS #1833: Checking error in the logs for points hierarchy

grzesiekb, 2021-07-30:
**Checking error in the logs for points hierarchy**

2. ERROR 2021-07-28T17:33:18,691 (org.scada_lts.dao.ViewHierarchyDAO.getNode:139) - org.scada_lts.exception.ViewHierarchyDaoException: org.springframework.jdbc.BadSqlGrammarException: PreparedStatementCallback; bad SQL grammar [call prc_vie
ws_hierarchy_select_node(?);]; nested exception is com.mysql.jdbc.exceptions.jdbc4.MySQLSyntaxErrorException: PROCEDURE scadaprod_706ded3_n.prc_views_hierarchy_select_node does not exist

grzesiekb, 2025-08-30:
Cześć, nie mam uprawnień, aby samodzielnie odpiąć się od tego zadania. Proszę o usunięcie mnie z listy Assignees. Dzięki!
Hi, I don’t have the permissions to unassign myself from this issue. Could you please remove me from the Assignees list? Thanks!

Limraj, 2025-08-31:
Currently, all migrations end with the creation of this procedure as well.