SCADA-LTS/Scada-LTS #1837: Python API

JoseVinicius1998, 2021-08-02:
Hello guys. Scada-LTS allows API with Python 3? Someone can help me? What module is more easy to use (suds, soappy, others)? Someone has an example of how to do? Thanks for the attention?

Sorry my english, i'm brazilian

radek2s, 2021-08-02:
Hello @JoseVinicius1998 
We are not sure what do you mean with "Scada-LTS allows API with Python 3". 
We have prepared Java REST API that allows user to request the application from any tool that supports HTTP Requests. We use in our purpose JavaScript for our interface but you can also use Python scripts to receive data from Scada-LTS. All supported methods are listed in our [API documentation file](https://github.com/SCADA-LTS/Scada-LTS/tree/develop/doc/RESTAPI). But if you want just to send a request for a data from Scada-LTS from your own Python script find the solutions on the Web how to use python request library. Example is provided here:

```python
import requests

url = 'http://localhost:8080/ScadaBR/api/point_value/getValue/{xid}'

x = requests.get(url)

print(x.text)
```

JoseVinicius1998, 2021-08-03:
I always get stuck because the code asks for authentication, both on soap and on Rest.
But I believe it can be solved. Thanks!

radek2s, 2021-08-03:
Ok so try to login using API:
```
http://localhost:8080/ScadaBR/api/auth/<<username>>/<<password>>
```
Just send that one request on the beginning, before you send request for data.

JoseVinicius1998, 2021-08-04:
Thanks Man!!! I did it! I am very grateful, the API is working perfectly! It's a big resource. I leave the code I used:

##############################################################
import requests

session = requests.Session()
url = 'http://localhost:8080/ScadaBR/api/auth/admin/admin'
x = session.get(url)
print(x.text)
url = 'http://scada.cagepa.pb.gov.br/ScadaBR/api/point_value/getValue/DP_722633'
y = session.get(url)
print(y.text)
##############################################################

JoseVinicius1998, 2021-08-04:
Thanks Man!!! I did it! I am very grateful, the API is working perfectly! It's a big resource. I leave the code I used:

##############################################################
import requests

session = requests.Session()
url = 'http://localhost:8080/ScadaBR/api/auth/admin/admin'
x = session.get(url)
print(x.text)
url = 'http://localhost:8080/ScadaBR/api/point_value/getValue/DP_722633'
y = session.get(url)
print(y.text)
##############################################################