SCADA-LTS/Scada-LTS #1855: what is different of dock image tag: v2.6.9 and tag: release-2.6.9

IndustrialCommunication, 2021-08-25:
what is different of dock image tag: v2.6.9 and tag: release-2.6.9,i' m  confused。

radek2s, 2021-08-25:
At that moment there should be the same because we have closed works on 'release/2.6.9'  branch but in other cases the Docker tag 'release-2.6.9' can contain features that should be tested. It is image for testing if everything is ok. Then if everyting is ok we publish a new Release of Scada-LTS and due to that GitHub Action is triggered to build image with official tag 'v2.6.9' in that case. 

Finally this 2 images should be the same but the tag 'vX.X.X' can be treat as official release of Scada-LTS and 'release-X.X.X' is the beta version of Scada for searching bugs and so on. 

I hope that it is understandable now, but you are right we should add some description about that to inform users about how it is working.

IndustrialCommunication, 2021-11-04:
i understand, release is one times and don't update in future, and vx.x.x is many times bug fix and renew.

closed 2021-11-04 by IndustrialCommunication