SCADA-LTS/Scada-LTS #1947: Scada-LTS installation on Windows without Docker

rodrigoms2004, 2021-11-03:
Folks,

I am  in a trouble to install Scada-LTS from scratch on Windows. 

When I run the catalina script I get this error

I am using [Tomcat 9](https://tomcat.apache.org/download-90.cgi), JDK-11, Mysql 5.7 and [Scada-LTS 2.6.10 WAR file] (https://github.com/SCADA-LTS/Scada-LTS/releases). 

Initially  I've used JDK-11 from this source http://jdk.java.net/archive and got an error. Then I got JDK from Scada-LTS documentation, same error.
https://github.com/SCADA-LTS/Scada-LTS/wiki/Getting-started---Development#java
https://jdk.java.net/java-se-ri/11
[image]

Then, I've took a look inside a Linux Docker Image that is working 
`root@b4ce1672ba27:/usr/local/tomcat# java -version
openjdk version "11.0.12" 2021-07-20
OpenJDK Runtime Environment 18.9 (build 11.0.12+7)
OpenJDK 64-Bit Server VM 18.9 (build 11.0.12+7, mixed mode, sharing)
root@b4ce1672ba27:/usr/local/tomcat# javac -version
javac 11.0.12
`
And put the same version from redhat site
https://developers.redhat.com/products/openjdk/download
[image]

And got the same error again...

**General configuration**

Catalina correctly searches by JAVA_HOME throught file _C:\Program Files\SCADA-LTS\bin\setenv.bat_
`set JAVA_HOME=C:\Program Files\SCADA-LTS\java\jdk-11`

The JDBC connector used is in [mysql-connector-java-5.1.49.jar](https://github.com/SCADA-LTS/Scada-LTS/raw/develop/WebContent/WEB-INF/lib/mysql-connector-java-5.1.49.jar) and it was put inside _C:\Program Files\SCADA-LTS\lib_ folder.

The _C:\Program Files\SCADA-LTS\conf\context.xml_ configuration model is [context.xml](https://github.com/SCADA-LTS/Scada-LTS/blob/develop/docker/config/context.xml), but with other DB parameters.
          _username="scadadbuser"
          password="scadapass"
          driverClassName="com.mysql.jdbc.Driver"
          defaultTransactionIsolation="READ_COMMITTED"
          connectionProperties="useSSL=false"
          url="jdbc:mysql://localhost:3306/scadalts"_

The tables in DB were correctly created by Scada-LTS during catalina initialization.

**Error logs**
[scadaLTSDAO.log](https://github.com/SCADA-LTS/Scada-LTS/files/7465251/scadaLTSDAO.log)
[mango.log](https://github.com/SCADA-LTS/Scada-LTS/files/7465252/mango.log)
[localhost.2021-11-03.log](https://github.com/SCADA-LTS/Scada-LTS/files/7465253/localhost.2021-11-03.log)
[flyway.log](https://github.com/SCADA-LTS/Scada-LTS/files/7465254/flyway.log)
[catalina.2021-11-03.log](https://github.com/SCADA-LTS/Scada-LTS/files/7465255/catalina.2021-11-03.log)

**The error in the catalina log**

_cataliana.bat run_

_03-Nov-2021 00:31:10.005 INFORMAÇÕES [mysql-cj-abandoned-connection-cleanup] org.apache.catalina.loader.WebappClassLoaderBase.checkStateForResourceLoading Illegal access: this web application instance has been stopped already. Could not load []. The following stack trace is thrown for debugging purposes as well as to attempt to terminate the thread which caused the illegal access.
	java.lang.IllegalStateException: Illegal access: this web application instance has been stopped already. Could not load []. The following stack trace is thrown for debugging purposes as well as to attempt to terminate the thread which caused the illegal access.
		at org.apache.catalina.loader.WebappClassLoaderBase.checkStateForResourceLoading(WebappClassLoaderBase.java:1427)
		at org.apache.catalina.loader.WebappClassLoaderBase.getResource(WebappClassLoaderBase.java:1052)
		at com.mysql.jdbc.AbandonedConnectionCleanupThread.checkThreadContextClassLoader(AbandonedConnectionCleanupThread.java:112)
		at com.mysql.jdbc.AbandonedConnectionCleanupThread.run(AbandonedConnectionCleanupThread.java:79)
		at java.base/java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1128)
		at java.base/java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:628)
		at java.base/java.lang.Thread.run(Thread.java:829)
03-Nov-2021 00:33:25.320 INFORMAÇÕES [http-nio-8080-exec-3] org.apache.jasper.servlet.TldScanner.scanJars At least one JAR was scanned for TLDs yet contained no TLDs. Enable debug logging for this logger for a complete list of JARs that were scanned but no TLDs were found in them. Skipping unneeded JARs during scanning can improve startup time and JSP compilation time.
03-Nov-2021 00:33:49.500 GRAVE [http-nio-8080-exec-3] org.apache.catalina.core.StandardContext.startInternal One or more listeners failed to start. Full details will be found in the appropriate container log file
03-Nov-2021 00:33:49.502 GRAVE [http-nio-8080-exec-3] org.apache.catalina.core.StandardContext.startInternal Context [/Scada-LTS] startup failed due to previous errors
03-Nov-2021 00:33:49.604 ADVERTÊNCIA [http-nio-8080-exec-3] org.apache.catalina.loader.WebappClassLoaderBase.clearReferencesThreads The web application [Scada-LTS] appears to have started a thread named [Log4j2-TF-2-Scheduled-1] but has failed to stop it. This is very likely to create a memory leak. Stack trace of thread:
 java.base@11.0.12/jdk.internal.misc.Unsafe.park(Native Method)
 java.base@11.0.12/java.util.concurrent.locks.LockSupport.parkNanos(LockSupport.java:234)
[... 56 lines cut ...]
 java.base@11.0.12/java.lang.Object.wait(Native Method)
 java.base@11.0.12/java.lang.Object.wait(Object.java:328)
 com.serotonin.timer.TimerThread.mainLoop(TimerThread.java:79)
 com.serotonin.timer.TimerThread.run(TimerThread.java:37)
03-Nov-2021 00:33:49.627 ADVERTÊNCIA [http-nio-8080-exec-3] org.apache.catalina.loader.WebappClassLoaderBase.clearReferencesThreads The web application [Scada-LTS] appears to have started a thread named [Image Fetcher 3] but has failed to stop it. This is very likely to create a memory leak. Stack trace of thread:
 java.base@11.0.12/java.lang.Object.wait(Native Method)
 java.desktop@11.0.12/sun.awt.image.ImageFetcher.nextImage(ImageFetcher.java:154)
 java.desktop@11.0.12/sun.awt.image.ImageFetcher.fetchloop(ImageFetcher.java:207)
 java.desktop@11.0.12/sun.awt.image.ImageFetcher.run(ImageFetcher.java:176)
03-Nov-2021 00:33:49.629 ADVERTÊNCIA [http-nio-8080-exec-3] org.apache.catalina.loader.WebappClassLoaderBase.clearReferencesThreads The web application [Scada-LTS] appears to have started a thread named [mysql-cj-abandoned-connection-cleanup] but has failed to stop it. This is very likely to create a memory leak. Stack trace of thread:
 java.base@11.0.12/java.lang.Object.wait(Native Method)
 java.base@11.0.12/java.lang.ref.ReferenceQueue.remove(ReferenceQueue.java:155)
 com.mysql.jdbc.AbandonedConnectionCleanupThread.run(AbandonedConnectionCleanupThread.java:80)
 java.base@11.0.12/java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1128)
 java.base@11.0.12/java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:628)
 java.base@11.0.12/java.lang.Thread.run(Thread.java:829)
03-Nov-2021 00:33:54.605 INFORMAÇÕES [mysql-cj-abandoned-connection-cleanup] org.apache.catalina.loader.WebappClassLoaderBase.checkStateForResourceLoading Illegal access: this web application instance has been stopped already. Could not load []. The following stack trace is thrown for debugging purposes as well as to attempt to terminate the thread which caused the illegal access.
	java.lang.IllegalStateException: Illegal access: this web application instance has been stopped already. Could not load []. The following stack trace is thrown for debugging purposes as well as to attempt to terminate the thread which caused the illegal access.
		at org.apache.catalina.loader.WebappClassLoaderBase.checkStateForResourceLoading(WebappClassLoaderBase.java:1427)
		at org.apache.catalina.loader.WebappClassLoaderBase.getResource(WebappClassLoaderBase.java:1052)
		at com.mysql.jdbc.AbandonedConnectionCleanupThread.checkThreadContextClassLoader(AbandonedConnectionCleanupThread.java:112)
		at com.mysql.jdbc.AbandonedConnectionCleanupThread.run(AbandonedConnectionCleanupThread.java:79)
		at java.base/java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1128)
		at java.base/java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:628)
		at java.base/java.lang.Thread.run(Thread.java:829)_

Limraj, 2021-11-03:
Hi!
Looks like {tomcat_home}/lib is missing library:

[tomcat_lib](https://github.com/SCADA-LTS/Scada-LTS/tree/develop/tomcat/lib)
and it is important that the connector is also in tomcat libs: mysql-connector-java-5.1.49.jar 

We need to update it in the documentation and in the script creating the docker image. Thanks for paying attention and a detailed description of the problem. If you are still having problems, let me know.

rodrigoms2004, 2021-11-04:
> Hi! Looks like {tomcat_home}/lib is missing library:
>
> [tomcat_lib](https://github.com/SCADA-LTS/Scada-LTS/tree/develop/tomcat/lib) and it is important that the connector is also in tomcat libs: [...]

rodrigoms2004, 2021-11-04:
Hi

I added this four files to tomcat/lib, and the same error occurred 

**In catalina log** 

_03-Nov-2021 23:42:44.560 INFORMAÇÕES [http-nio-8080-exec-6] org.apache.jasper.servlet.TldScanner.scanJars At least one JAR was scanned for TLDs yet contained no TLDs. Enable debug logging for this logger for a complete list of JARs that were scanned but no TLDs were found in them. Skipping unneeded JARs during scanning can improve startup time and JSP compilation time.
03-Nov-2021 23:42:46.847 GRAVE [http-nio-8080-exec-6] org.apache.catalina.core.StandardContext.startInternal One or more listeners failed to start. Full details will be found in the appropriate container log file
03-Nov-2021 23:42:46.850 GRAVE [http-nio-8080-exec-6] org.apache.catalina.core.StandardContext.startInternal Context [/Scada-LTS] startup failed due to previous errors
03-Nov-2021 23:42:46.956 ADVERTÊNCIA [http-nio-8080-exec-6] org.apache.catalina.loader.WebappClassLoaderBase.clearReferencesThreads The web application [Scada-LTS] appears to have started a thread named [Log4j2-TF-2-Scheduled-1] but has failed to stop it. This is very likely to create a memory leak. Stack trace of thread:
 java.base@11.0.12/jdk.internal.misc.Unsafe.park(Native Method)
 java.base@11.0.12/java.util.concurrent.locks.LockSupport.parkNanos(LockSupport.java:234)
 java.base@11.0.12/java.util.concurrent.locks.AbstractQueuedSynchronizer$ConditionObject.awaitNanos(AbstractQueuedSynchronizer.java:2123)
 java.base@11.0.12/java.util.concurrent.ScheduledThreadPoolExecutor$DelayedWorkQueue.take(ScheduledThreadPoolExecutor.java:1182)
 java.base@11.0.12/java.util.concurrent.ScheduledThreadPoolExecutor$DelayedWorkQueue.take(ScheduledThreadPoolExecutor.java:899)
 java.base@11.0.12/java.util.concurrent.ThreadPoolExecutor.getTask(ThreadPoolExecutor.java:1054)
 java.base@11.0.12/java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1114)
 java.base@11.0.12/java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:628)
 java.base@11.0.12/java.lang.Thread.run(Thread.java:829)
03-Nov-2021 23:42:46.980 ADVERTÊNCIA [http-nio-8080-exec-6] org.apache.catalina.loader.WebappClassLoaderBase.clearReferencesThreads The web application [Scada-LTS] appears to have started a thread named [AsyncAppender-ASYNC_MODBUS] but has failed to stop it. This is very likely to create a memory leak. Stack trace of thread:
 java.base@11.0.12/jdk.internal.misc.Unsafe.park(Native Method)
 java.base@11.0.12/java.util.concurrent.locks.LockSupport.park(LockSupport.java:194)
 java.base@11.0.12/java.util.concurrent.locks.AbstractQueuedSynchronizer$ConditionObject.await(AbstractQueuedSynchronizer.java:2081)
 java.base@11.0.12/java.util.concurrent.ArrayBlockingQueue.take(ArrayBlockingQueue.java:417)
 org.apache.logging.log4j.core.appender.AsyncAppender$AsyncThread.run(AsyncAppender.java:398)
03-Nov-2021 23:42:46.985 ADVERTÊNCIA [http-nio-8080-exec-6] org.apache.catalina.loader.WebappClassLoaderBase.clearReferencesThreads The web application [Scada-LTS] appears to have started a thread named [AsyncAppender-ASYNC_SCADA-LTS_DAO] but has failed to stop it. This is very likely to create a memory leak. Stack trace of thread:
 java.base@11.0.12/jdk.internal.misc.Unsafe.park(Native Method)
 java.base@11.0.12/java.util.concurrent.locks.LockSupport.park(LockSupport.java:194)
 java.base@11.0.12/java.util.concurrent.locks.AbstractQueuedSynchronizer$ConditionObject.await(AbstractQueuedSynchronizer.java:2081)
 java.base@11.0.12/java.util.concurrent.ArrayBlockingQueue.take(ArrayBlockingQueue.java:417)
 org.apache.logging.log4j.core.appender.AsyncAppender$AsyncThread.run(AsyncAppender.java:398)
03-Nov-2021 23:42:46.990 ADVERTÊNCIA [http-nio-8080-exec-6] org.apache.catalina.loader.WebappClassLoaderBase.clearReferencesThreads The web application [Scada-LTS] appears to have started a thread named [AsyncAppender-ASYNC] but has failed to stop it. This is very likely to create a memory leak. Stack trace of thread:
 java.base@11.0.12/jdk.internal.misc.Unsafe.park(Native Method)
 java.base@11.0.12/java.util.concurrent.locks.LockSupport.park(LockSupport.java:194)
 java.base@11.0.12/java.util.concurrent.locks.AbstractQueuedSynchronizer$ConditionObject.await(AbstractQueuedSynchronizer.java:2081)
 java.base@11.0.12/java.util.concurrent.ArrayBlockingQueue.take(ArrayBlockingQueue.java:417)
 org.apache.logging.log4j.core.appender.AsyncAppender$AsyncThread.run(AsyncAppender.java:398)
03-Nov-2021 23:42:47.017 ADVERTÊNCIA [http-nio-8080-exec-6] org.apache.catalina.loader.WebappClassLoaderBase.clearReferencesThreads The web application [Scada-LTS] appears to have started a thread named [AsyncAppender-ASYNC_DAO] but has failed to stop it. This is very likely to create a memory leak. Stack trace of thread:
 java.base@11.0.12/jdk.internal.misc.Unsafe.park(Native Method)
 java.base@11.0.12/java.util.concurrent.locks.LockSupport.park(LockSupport.java:194)
 java.base@11.0.12/java.util.concurrent.locks.AbstractQueuedSynchronizer$ConditionObject.await(AbstractQueuedSynchronizer.java:2081)
 java.base@11.0.12/java.util.concurrent.ArrayBlockingQueue.take(ArrayBlockingQueue.java:417)
 org.apache.logging.log4j.core.appender.AsyncAppender$AsyncThread.run(AsyncAppender.java:398)
03-Nov-2021 23:42:47.022 ADVERTÊNCIA [http-nio-8080-exec-6] org.apache.catalina.loader.WebappClassLoaderBase.clearReferencesThreads The web application [Scada-LTS] appears to have started a thread named [AsyncAppender-ASYNC_FLYWAY] but has faile [... cut ...]
[localhost_access_log.2021-11-02.txt](https://github.com/SCADA-LTS/Scada-LTS/files/7471978/localhost_access_log.2021-11-02.txt)
[catalina.2021-11-02.log](https://github.com/SCADA-LTS/Scada-LTS/files/7471979/catalina.2021-11-02.log)
[catalina.2021-11-03.log](https://github.com/SCADA-LTS/Scada-LTS/files/7471980/catalina.2021-11-03.log)
[localhost.2021-11-02.log](https://github.com/SCADA-LTS/Scada-LTS/files/7471981/localhost.2021-11-02.log)
[localhost.2021-11-03.log](https://github.com/SCADA-LTS/Scada-LTS/files/7471982/localhost.2021-11-03.log)
[mango.log](https://github.com/SCADA-LTS/Scada-LTS/files/7471983/mango.log)

grzesiekb, 2021-11-15:
The installer for Scada-LTS is being developed and is not yet finished.

KMKAR, 2021-11-18:
Hi @rodrigoms2004 / @grzesiekb 
I installed the latest 2.6.1 war release in a fresh install of raspbian 11 (rpi4)
Application is failing to start on tomcat and the scadaLTSDAO.log file shows the same in my case:
`ERROR 2021-11-18T11:18:13,081 (?.:) - org.springframework.jdbc.BadSqlGrammarException: PreparedStatementCallback; bad SQL grammar []; nested exception is com.mysql.jdbc.exceptions.jdbc4.MySQLSyntaxErrorException: Unknown column 'shortMessage' in 'field list'`

Do you know how to fix this issue? Thanks

rodrigoms2004, 2021-11-19:
> The installer for Scada-LTS is being developed and is not yet finished. https://github.com/SCADA-LTS/installer. The current version does not finished for windows.

@KMKAR  our colleague @grzesiekb had the same issue in linux platform, maybe it could be a DB issue

celsou, 2021-11-29:
Hello everyone,

I tried to install on Linux and got the same error as @KMKAR. I've been inspecting the database tables and found that there was no "shortMessage" column in my database.

My guess is that this and other database modifications are performed by [these classes](https://github.com/SCADA-LTS/Scada-LTS/tree/3fc39ea30c3d194ccd5505c54dd0ce1ca470e817/src/org/scada_lts/dao/migration/ mysql), but for some reason I don't know they aren't running during manual installation.

I hope this info can help.

grzesiekb, 2021-11-29:
Feel free to test and comment on the installer - It downloads a set of matching components like a particular version of tomcat java, downloads the appropriate libraries and configures tomcat and downloads Scada-LTS.war which should work in that environment. This assumes you have mysql 5.7 installed. 
https://github.com/SCADA-LTS/installer/releases

you can try to use inst_linux and at the end of the file start.sh

1. download the inst_linux file into a separate directory 
2. download the file start.sh 
3. Run inst_linux -h 
4. If we have a default configuration: root/root on localhost 3306 with scadalts database, it runs inst_linux
5. If you have a different configuration you have to provide it in the proper arguments for inst_linux which you can see by running inst_linux -h
6. When everything is downloaded and configured run start.sh file (you may need to give it permissions to execute)

Translated with www.DeepL.com/Translator (free version)

Limraj, 2021-11-29:
Hi, 
sorry that I didn't reply, for some reason I didn't get notifications about the update of this topic. 

Try to do what @grzesiekb wrote, but the problems in your installation seem to be not including the appropriate libs in tomcat/lib if you want to come back to this, try again:

1. In addition to these jars from the link, you also need to put a connector to the database: mysql-connector-java-5.1.49.jar (tomcat/lib)
2. If these jars are in this location, you'd have to run the application on a clean database.
If the default database does not contain data, you can do it based on:
drop database sacadalts;
create database scadalts;
and run application;

This should help looking at the attached logs. (now there are no bugs related to missing libs)
Regards and sorry again for the late reaction.

celsou, 2021-11-29:
Good news!

Deleting the "scadalts" database and recreating it solved the problem and I was able to run Scada-LTS. Now I'm going to download the installer in Rust and try it out so I can give you feedback.

jrajnhold, 2022-01-19:
Hi Everyone!
I try to run ScadaLTS on ubuntu but at the end I have the same problem as you all:
_ERROR 2021-11-18T11:18:13,081 (?.:) - org.springframework.jdbc.BadSqlGrammarException: PreparedStatementCallback; bad SQL grammar []; nested exception is com.mysql.jdbc.exceptions.jdbc4.MySQLSyntaxErrorException: Unknown column 'shortMessage' in 'field list'
How to fix it to check the new version of SCADA-LTS?

PS. I try to drop and create database - it's not helping!

Limraj, 2022-11-01:
Hello @jrajnhold, 
Have you tried on newer versions or does the problem occur? This error shows that he did not perform the migration, hence the column that is created in a certain version of the migration is missing.

closed 2023-03-26 by Limraj