SCADA-LTS/Scada-LTS #1959: Who to contact for security issues

zidingz, 2021-11-24:
Hey there!

I belong to an open source security research community, and a member (@h3v0x) has found an issue, but doesn’t know the best way to disclose it.

If not a hassle, might you kindly add a `SECURITY.md` file with an email, or another contact method? GitHub [recommends](https://docs.github.com/en/code-security/getting-started/adding-a-security-policy-to-your-repository) this best practice to ensure security issues are responsibly disclosed, and it would serve as a simple instruction for security researchers in the future.

Thank you for your consideration, and I look forward to hearing from you!

(cc @huntr-helper)

SoftQ, 2021-11-25:
Hello!
Thanks for contacting us. Please write the email to scadalts-support@softq.pl.
Thanks

closed 2021-11-25 by SoftQ

JamieSlome, 2021-11-25:
@SoftQ - e-mail sent! 🎉