SCADA-LTS/Scada-LTS #1961: Graphical views Scada-LTS

jmuguruza, 2021-11-25:
I am running OpenPLC (port 8081) and ScadaLTS (port 8080 with mysql database) using Docker Compose. 
I loaded an example program on the PLC, taken from GitHub, named waterheater.st and then ran the PLC. 
So, now I make the connection with Scada-LTS using Modbus IP on port 502, after including the background image of the waterheater program and the .json file at the import/export.
Then I go to graphical views and try to interact with some buttons that I create, but I got an error from localhost:8080 saying:
Incorrect result size: expected 1, actual 0

Can someone find a solution for this? 
Thanks in advance

Limraj, 2021-11-25:
Hello @jmuguruza, 
some questions:
1. When you edit data source Modbus IP, do you have any errors in:
[image]
Perhaps you could make a screenshot?
2. Can we get the json file you are trying to import? (kamil.jarmusik@softq.pl) 
3. You import via the function Import / Export-> paste json into the "data" text field and click the "Import" button? (marked in red in the screen) [image]
4. Maybe you would be able to make a screenshot when this error occurs?

Regards

jmuguruza, 2021-11-29:
1- I do not get any errors
3- I copy the whole file and paste it into the "data" field and click "Import" right, and I get the same errors. 
[image]
4- I get an error like the one of the image below
[image]

closed 2022-01-08 by jmuguruza