SCADA-LTS/Scada-LTS #2005: Fixed permission for import/export project

Limraj, 2021-12-20:
**Description**
The project import function is partially secured, but clicking by the user leads to the first stage of importing the project. Full security for the import function should be added.

**To Reproduce**
Steps to reproduce the behavior:
1. Create user as usr_1;
2. Login on usr_1;
2. Go to '/ScadaBR/emport.shtm'
3. In section 'Import Project (Upload)' click on 'Choose file' select file and click 'Send'
4. Click 'Import' -> unknown error

**Specification:**
 - Version 2.6.10

grzesiekb, 2021-12-30:
After switching to the branch with this task, I get an error:
Exception in thread "pool-3-thread-16" java.lang.NullPointerException
	at com.serotonin.mango.rt.maint.WorkItemMonitor.run(WorkItemMonitor.java:45)
	at com.serotonin.timer.TimerTask.run(TimerTask.java:143)
	at java.base/java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1128)
	at java.base/java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:628)
	at java.base/java.lang.Thread.run(Thread.java:829)

grzesiekb, 2021-12-30:
Next error:
Exception in thread "pool-3-thread-16" java.lang.NullPointerException
	at com.serotonin.mango.rt.maint.WorkItemMonitor.run(WorkItemMonitor.java:45)
	at com.serotonin.timer.TimerTask.run(TimerTask.java:143)
	at java.base/java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1128)
	at java.base/java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:628)
	at java.base/java.lang.Thread.run(Thread.java:829)

closed 2022-01-18 by Limraj