SCADA-LTS/Scada-LTS #2014: [bug] Point Hierarchy not loading after import

radek2s, 2021-12-24:
**Describe the bug**
The Data Point Hierarchy is not initialized on the classic UI and on the new UI after importing the structure from Import/Export menu. But after manipulation with that object it is becoming visible. 

**To Reproduce**

1. Create a Point Hierarchy structure. 
2. Then try to export it.
3. Delete everything in Point Hierarchy.
4. Import point hierarchy structure.
5. Now you can see that the structure has not been loaded.

But:
After adding a new directory user will notice that the structure is loaded.
So it is imported but it is not initialized and that may be the reason why the point hierarchy is not visible.

**Expected behavior**
After Import/Export the Point Hierarchy should be working

**Desktop (please complete the following information):**
 - OS: Linux
 - Browser Chrome
 - Version 93

grzesiekb, 2021-12-29:
I understand that this concerns the new hierarchy.
e.g. on event on active the data should be updated and you can try to force only one window to open with editing of the score hierarchy

grzesiekb, 2021-12-29:
In the old tab there was an assumption that the user configuring it is an aware user and works on one tab and if he has two open he can refresh them.
The bigger problem is the speed of loading data and its correct rendering.

Limraj, 2023-09-11:
I looked at the code and it looks like the structure is updated every 10min, but the refresh button should force this update...

The refresh rate is configurable in the env.properties file:
abilit.CRONE_UPDATE_CACHE_POINT_HIERARCHY=0 0/10 * * * ?

And fixed button refresh: #2484