SCADA-LTS/Scada-LTS #2019: Fixed 'View charts' for report with events

Limraj, 2021-12-29:
**Description**
For a report with events, an attempt to generate charts ends with an exception "System Exception";

**To Reproduce**
1. Create datasource ds_1 (virtual, enable), datapoint dp_1 (settable, numeric, no change), eventdetector ed_1 (change, information);
2. Go to 'Watchlists' and change value for dp_1, this should generate an event with the information level;
3. Go to 'Reports';
4. Click on 'New report';
5. Set rep_1 in 'Report name';
6. Add point ds_1 - dp_1 in 'Points';
7. Select 'All events' in 'Events'
8. Check 'Relative to report time'-> 'Past';
9. Click save;
10. Click 'Run now', and reload page;
11. Report rep_1 should appear in 'Report queue' table;
12. Click on 'View charts' for report rep_1 -> System Exception;

**Specification:**
 - Scada version: 2.6.10

closed 2022-01-18 by Limraj

Golffies, 2022-01-20:
The "system exception" error was not reproducible, neither on version 2.6.1 (scadadev), nor on version 2.6.10-rc1 (scadatest).

In both cases, it is possible to display the charts or export the data without triggering the error.

In attachment, the display of the charts resulting from the test on 2.6.10-rc1 on the left and on 2.6.1 on the right, without error.

[image]
.

Golffies, 2022-01-20:
The "system exception" error was reproducible on version 2.6.1 (scadadev - on the right), provided that the new report is applied to a data point with a "high limit" type event detector. Under that same condition, the same error was not reproducible on version 2.6.10-rc1 (scadatest - on the left).

[image]