SCADA-LTS/Scada-LTS #2021: ScadaLTS don't work with MariaDB 10

LinuxJohannes, 2021-12-29:
**ScadaLTS don't work with MariaDB 10**

Did a setup on an Raspberry PI with a current, new updated Raspbian image. 
Installed is:

- Tomcat = 9.0.31 (Debian)
- ScadaLTS = War-File v2.6.10
- Java JDK = 11.0.12
- JVM = 11.0.12+7-post-Raspbian-2deb10u1
- MariaDB = MySQL 5.5.5-**10.0.28**-MariaDB-2+b1
- Flyway = 7 < got it from Log

Tomcat can't start ScadaLTS. Found in /var/log/catalina.out this errormessage:

ERROR 2021-12-29T09:18:12,801 (com.serotonin.mango.db.MySQLAccess.newDatabaseCheck:157) - org.flywaydb.core.internal.license.FlywayEditionUpgradeRequiredException: Flyway Enterprise Edition or MariaDB upgrade required: MariaDB 10.0 is no longer supported by Flyway Community Edition, but still supported by Flyway Enterprise Edition.

SoftQ, 2021-12-29:
Hello
We don't have any influence on Flyway policy of supported database versions.. Here are two ideas 

1.  As the log states, please try again on a newer version of MariaDB like 10.3.  Here You can find a tutorial that shows MariaDB 10.3 on RPi with Debian 10:

https://r00t4bl3.com/post/how-to-install-mariadb-server-on-debian-10-buster

2. Other possible (not verified) hack&play idea would be to take backup of empty Scada-LTS database on other machine, and restore it to Your older MariaDB.. 
- some minor adaptations of backup SQL file might be required
- despite Flyway would not have to perform any migrations it may prevent Scada from running (let us know then !)
- some specific MariaDB version may not work well with Scada.. 

Good luck, please update us on progress!

LinuxJohannes, 2021-12-29:
Solved. 
Many thanks for your help.

Followed your link 
     
https://r00t4bl3.com/post/how-to-install-mariadb-server-on-debian-10-buster

for upgrade to MySQL 5.5.5-**10.3.31**-MariaDB-0+deb10u1

closed 2021-12-29 by LinuxJohannes