SCADA-LTS/Scada-LTS #2024: Docker deployment: SSL certbot from Let's Encrypt

hjbleier, 2021-12-30:
**Problem**
Since the recommended way to deploy Scada-LTS is now to use docker-compose, I have not been able to figure out how to setup and automate the update of Let's Encrypt certbot on my deployment of Scada-LTS 2.6.10. This is causing some problems with some mobile and desktop browsers versions that don't allow the user to override and force the browser to connect over regular http and ignore https. Some of the users do not have administrative right to the devices so it is not an option to install a browser that does not enforce https everywhere. This is causing problems with the users not to be able to test the Scada-LTS deployment on their devices.

**Solution**
Can someone with more docker experience add a page to the project wiki on the steps to accomplish this. This I am sure will help many people out. The amount of browsers that are requiring https for all pages is growing fast and if you don't have administrative rights to the device your can't bypass this in many situations.

**Hosting System**
Host system: Ubuntu Linux Server 20.04.3 LTS
Running docker image with the detach "-d" option to retain the project data.
system init scripts to auto launch the docker containers for database and Scada-LTS on boot.
Other than the above modifications it is a fresh install of Ubuntu that is dedicated to Scada-LTS.

SoftQ, 2021-12-30:
Hello
The following may work for You
https://mvysny.github.io/Lets-Encrypt-with-Docker-Tomcat
Let us know!

Limraj, 2021-12-30:
As for running scada with docker-compose, we have in the wiki: https://github.com/SCADA-LTS/Scada-LTS/wiki

Based on the instructions for release 2.6.10:
1) Copy the contents of the docker-compose.yml file from the main application source directory: docker-compose.yml
2) Change docker tag from master to release:
image: scadalts/scadalts:master
on
image: scadalts/scadalts:release-2.6.10
3) Run docker, make sure that we do not have a locally running database or other applications on ports: 3306, 8080, then go to the directory with the docker-compose.yml file and run in the following order:
* docker-compose database up;
* docker-compose scadalts up;

I am attaching the modified docker-compose.yml:
[docker-compose.yml.zip](https://github.com/SCADA-LTS/Scada-LTS/files/7793112/docker-compose.yml.zip)

Here is a list of the available tags with the app:
https://hub.docker.com/r/scadalts/scadalts

hjbleier, 2022-01-18:
> Hello The following may work for You https://mvysny.github.io/Lets-Encrypt-with-Docker-Tomcat Let us know!
> 
> PS. The Docker image is not a production-grade one. It is intended for development, evaluation and training purposes.

That link helped a lot thank you!

I didn't do the wiki docker-compose method, I built my own docker-compose.yml based on:

mysql/mysql-server:5.7
and
tomcat:9.0.53

then it copies the latest war release version of Scada-LTS into the tomcat docker.
I added the letsencrypt volume similar to what the link said to do and it all work perfect.

Thanks again for pointing me in the right direction

closed 2023-03-26 by Limraj