SCADA-LTS/Scada-LTS #2034: Showing scripts in graphical view

jmuguruza, 2022-01-08:
I have created a .json file and a background image in order to preview one part of a project.
I defined 9 different HTML data points and 9 scripts. Now I want to show an output value (arbitrary value of 100 decided by me) at the output box after "Caudal de Aire:". This integer value comes from one of the nine scripts defined at the .json file.
I am not able to see that value in graphical views; the script declaration at the file must be wrong.
[image]
[image]

Does anyone know how to define it in the .json file?

Thanks in advance

Limraj, 2022-01-10:
Hi @jmuguruza,
1. If you would like to return a constant value there, it is enough in the script:
"return 100";
2. If you would like to return there the current point value that is set in this component, the script will suffice:
"return this.value";
You also have access to other point data here, for example:
"return point.name + ''" or 
"return point.xid + ''" and other

jmuguruza, 2022-01-10:
All right. Thank you for your help @Limraj !!!

Limraj, 2022-01-10:
@jmuguruza No problem, do you have any other questions?

Limraj, 2022-01-10:
@jmuguruza Can I close this topic?

jmuguruza, 2022-01-10:
Yes, by the moment everything is ok.
Thanks!

closed 2022-01-10 by Limraj

jmuguruza, 2022-01-13:
Now I want to show at the inputs box an integer value from OpenPLC.
In the case of "Caudal de Purga", I gave it a value of 95145 in OpenPLC Editor and uploaded the .st file to OpenPLC.
At the .json file I defined the script like this:
"script":"return(\"<p style='font-size:25px'>\" + parseFloat(this.value).toFixed(0) + \"<\/p>\") ",

Is there something wrong or I might have made another mistake?

jmuguruza, 2022-01-13:
Now I want to show the variable "Caudal de Purga" at the input box; this variable come from OpenPLC.
I defined it as an integer in OpenPLC Editor, initialized with value 95145. 
So I would like to see that number at the Scada graphical views. 
I used this instruction at the .json file when defining the script:

"script":"return(\"<p style='font-size:25px'>\" + parseFloat(this.value).toFixed(0) + \"<\/p>\") "

Is there anything wrong or I might have made another mistake?

Limraj, 2022-01-13:
hello @jmuguruza,
The problem in your definition is that the whole property "script" is one string, the problem is with it + and using ", lower level denoted by `\"` (nesting), try: 
`"script":"return(parseFloat(this.value).toFixed(0) + \"<\/p>\");",`

jmuguruza, 2022-01-13:
I still got the same result; a 0.
[image]

Limraj, 2022-01-13:
@jmuguruza Could you please send me the entire system configuration by e-mail: kamil.jarmusik@softq.pl? (To do this, go to 'Import / Export' -> 'Export Project (Download)' section -> 'Export Project' button
1. As for the value 0:
try setting 
`return this.value `
there, although it will probably be 0. Another possible cause is something wrong with the connection. If you send me the configuration, it will be easier for me to help. You can also describe exactly in the email what you set up in OpenPLC.
2. As for the appearance of this value, there must be something wrong with the html / css styles structure in the view definition.

jmuguruza, 2022-01-16:
My situation now is the following:

1- I have created an .st file, with memory addressing that I did not do before and the monitor shows the following:
[image]

2- I changed a little bit the .json file, 'pointLocator' arguments at the 'dataPoints' part. These arguments are 'range', 'modbusDataType' and 'slaveID'.

3- I import the .json file at the 'Import/Export' menu and get this values:
[image]

Limraj, 2022-01-16:
Hello @jmuguruza 

I think it's a matter of configuring the correct registers, addresses, and types, in Scada you change the address using the offset parameter (0-65535), then you have 4 types of registers:
1) Coins status - read/write 1 bit register
2) Inputs status - read 1 bit register
3) Holding register - read/write 16 bit register
4) Input register - read 16 bit register

and what you changed modbusDataType to test yourself and determine these settings, type, register and addresses, you can use the section to test the connection:
[image]

Modbus documentation
https://modbus.org/docs/Modbus_Application_Protocol_V1_1b3.pdf

Greetings,
Kamil Jarmusik

jmuguruza, 2022-01-17:
Hi @Limraj 

I defined the 7 inputs as Input Registers (read) in the .json file and the 2 outputs as Holding Registers (read/write).

Now I don't know how to deal with the offset. How should I configure that parameter?

Limraj, 2022-01-17:
Hey @jmuguruza , 
each register type has the addresses 0-65535. See yourself for documentation, I also showed you how to test it in the Scada, you have to look into the program in which you configure points and look for something that may be responsible for it, in my opinion you can try extreme addresses, probably by default either addresses from 0-65535 (try 0, 1, 2, 3...) or 65535-0 (65535, 65534...), maybe it has to do with "location" or that .st file that you defined. 

Greetings,
Kamil Jarmusik

jmuguruza, 2022-01-17:
Fine, I think that I solved this problem.

Thank you once again @Limraj.

jmuguruza, 2022-01-23:
Hi @Limraj, one last thing. I achieved this in Scada-LTS:
[image]

I am able to see all the values in the graphical views, but can not receive those two output variables. They are defined this way:
[image]

Then the data points are defined as 2 Byte int signed holding registers and I get this in OpenPLC:
[image]

Limraj, 2022-01-23:
Hi, 
I don't quite understand, because in these scripts you return the constant values ​​100 and 57087, so how come you want the openplc value there? You need to change to this.value there, but I already wrote about that...

Greetings,
Kamil Jarmusik

jmuguruza, 2022-01-23:
The input variables come from OpenPLC and I use this.value and get those values, but now I am writing myself that output value of 100 and want to send it to OpenPLC. That is what I did not manage to do.

Limraj, 2022-01-23:
To enable the value to be changed:
1) The first shorter solution, you need to set the point as settable, read / write register, and create a 'simple point' component on the view with this point, and select the "Display controls" parameter in it. Then, after hovering over the point, you should be able to change the value. This solution is rather recommended in case there is a problem, let me know.

2) Another solution that requires more steps but uses scripts:
a) create datasource as ds_1 (virtual) and datapoint (numeric, settable, no change) as dp_to_read;
b) create script as script_write in the scripts tab, adding datapoint dp_to_read as val_read to the context and using a special function there
`val_2.writeDataPoint ('DP_OPENPLC_XID', val_read.value); `
[image]
c) then creating two components on the view for setting values, 'simple point' and setting dp_to_read in it, and component 'Button (script)' with script_write script to change the value.

Greetings,
Kamil Jarmusik

jmuguruza, 2022-01-23:
If I select one of the two output values and click on "Point details" then I can set a value manually and get it in the OpenPLC monitoring. It works with a value of 100, but when writing 57087 it does not work (both are defined as integers).
[image]
[image]

Limraj, 2022-01-23:
Everything I have told you so far should enable you to solve these types of problems. The other thing is that you don't listen to the instructions, so act your way. I can't help you any more. Greetings.

Limraj, 2022-01-25:
@jmuguruza 
Hey, did you deal with the problem?

jmuguruza, 2022-01-28:
For the last output variable (57087) no, the program did not let me set a value like that one.
But it was part of my final degree project, that I finished some days algo, so do not worry.
Thank you for your help this weeks :-)