SCADA-LTS/Scada-LTS #2041: do have the  plan to support mysql 8

IndustrialCommunication, 2022-01-18:
according to the mysql official announce，mysql 8 performance is more  better  then mysql 5, do have the plan to support?

Limraj, 2022-01-19:
Hey @IndustrialCommunication , we are considering a version upgrade, we are also working on a support version for QuestDB, for part of the system.

Greetings,
Kamil Jarmusik

fabiodurao, 2022-01-19:
Hey @Limraj I would like to understand what motivated the team to focus on using QuestDB. He's better? Faster? Uses less space than mysql?

IndustrialCommunication, 2022-01-20:
@fabiodurao QuestDB is more better than mysql in iot,they are not  a level.

Golffies, 2022-01-20:
In future versions of SCADA-LTS, QuestDB will not replace mySQL, but will be deployed in parallel. General settings, such as user rights for example, will be kept in mySQL. Data acquired from the field, typically from PLCs, will be stored in QuestDB.

QuestDB's superior performance in processing large volumes of time-stamped data would make it possible to use SCADA-LTS to track the history of measurements back years.

closed 2023-03-14 by Limraj