SCADA-LTS/Scada-LTS #2042: simplePoint changevalue popup has no background

hjbleier, 2022-01-18:
**Problem**
There is no background on the value change popup window.
It appears to be a problem with the CSS for the modern theme.

**To Reproduce**
Scada-LTS 2.6.10-rc1
Modern Theme selected for the user
drop a simplePoint on a new view display
check box to enable controls
then save the view
open up the new view page
try clicking on the value to change.

**Expected behavior**
There should be a white background or other color so the user
doesn't get confused when text overlaps.

**Environment**
 - OS: Linux
 - Browser: chrome, firefox
 - Version: Scada-LTS 2.6.10-rc
 
[image]
1

Limraj, 2022-01-19:
#2042

hjbleier, 2022-01-28:
@radek2s Thank you. I manually added your changes to my current deployment of 2.6.10-rc1 and it fixed the problem, just had to clear the web browsers cache on all the machines that previously logged in. Thanks!

closed 2022-03-22 by Limraj