SCADA-LTS/Scada-LTS #2044: how about this project commercial models? how to continue heathly

huanghe-1, 2022-01-18:
After all，engineer need food,company need to survive

Limraj, 2022-01-19:
Hey @huanghe-1 , if you have any suggestion, please contact: 
michal.rokitianski@softq.pl
kamil.jarmusik@softq.pl

Greetings,
Kamil Jarmusik

huanghe-1, 2022-01-19:
do your company have some Careers？i'm pleasure to join

Limraj, 2023-06-03:
Hi @huanghe-1,
If you still think about some kind of cooperation, e.g. you have some clients who need Scada, write to Fryderyk.
If you would like to get involved as a programmer/engineer, we encourage you to get involved in the project(e.g. proposing changes, corrections, testing our protocols, etc.), we'll see what we come up with.

Regards,
Kamil Jarmusik

closed 2023-06-03 by Limraj