SCADA-LTS/Scada-LTS #2045: Server error on user update

sselvaggi, 2022-01-19:
**To Reproduce**
1 Go to user list in new ui
2 edit  any user's property
3 save and check server is not persisting it (error 500)

Limraj, 2022-01-19:
In log:
`PreparedStatementCallback; SQL [update users set username=?, firstName=?, lastName=?, password=?, email=?, phone=?, admin=?, disabled=?, homeUrl=?, receiveAlarmEmails=?, receiveOwnAuditEvents=?, hideMenu=?, theme=? where id=? ]; Column 'password' cannot be null`

Limraj, 2022-01-19:
Scenario:

1) Create two new user as usr_1 and usr_2, and profile pro_1;

2) Update usr_1: 
a) "Change password" by admin and non-admin user;
b) "First name" by admin and non-admin user;
c) "Last name" by admin and non-admin user;
d) "Phone number" by admin and non-admin user;
e) "Receive alarm events" by admin and non-admin user;
f) "Receive alarm emails" by admin and non-admin user;
g) "Theme UI" by admin and non-admin user;
h) "User Profile" by admin for usr_1 set pro_1;

3) Delete usr_1, usr_2;

Limraj, 2022-02-18:
Has been merged into preview/2.7.1 and is close.

closed 2022-02-18 by Limraj