SCADA-LTS/Scada-LTS #2063: Email event handler overrides script event handler

grzesiekb, 2022-02-02:
**Describe the bug**
Email event handler overrides script event handler or event detectors disappear

**To Reproduce**
Steps to reproduce the behavior:
1. Create point in modbus ds with event detector, one - no update 5 seconds, second  change 
2. Create point in virtual data source DP_LOCALIZATION_PLC_IS_NOT_CHANGED (not change, setable, start value 0)
2. Create scripts:
a) PLC_IS_NOT_CHANGE 
```
if (p9801.value != 0) {  // point modbus 
  dp.writeDataPoint("DP_LOCALIZATION_PLC_IS_NOT_CHANGED", 1); // virutal data source
}
```
b) PLC_IS_NOT_CHANGE_RESET
```
dp.writeDataPoint("DP_LOCALIZATION_PLC_IS_NOT_CHANGED", 0); // virutal data source
```

3. In event handlers set 
a) for event detector "no change" - script PLC_IS_NOT_CHANGE
b) for event detector "change" - script PLC_IS_NOT_CHANGE_RESET

4. In event handlers set next event handler for event detector "no_change" - send email

**Expected behavior**
Correct action for event detector

**Problems**
1) When set second event handler (email) for event detector "no change" scipt PLC_IS_NOT_CHANGE is not working.
2) After some times event detector for point modbus is disappear

closed 2022-04-06 by Limraj

Limraj, 2023-01-14:
https://github.com/SCADA-LTS/Scada-LTS/issues/1465

Limraj, 2023-01-14:
#1069