SCADA-LTS/Scada-LTS #2132: Updated docker-compose.yml with CATALINA_OPTS

Limraj, 2022-03-11:
**Target**
1. Among other things, it allows you to set parameters InitialHeapSize and MaxHeapSize, and other options for JVM;
2. Change master on latest tag;

kinsamanka, 2022-03-14:
You can also add [wait-for-it.sh](https://github.com/vishnubob/wait-for-it) so that `scadalts` will start only once the database is ready to accept connections:

```
    scadalts:
        image: scadalts/scadalts:latest
        environment:
            - CATALINA_OPTS=-Xmx512m -Xms512m
        ports: 
            - "8080:8080"
        depends_on: 
            - database
        expose: ["8080", "8000"]
        links:
            - database:database
        command:
          - /usr/bin/wait-for-it.sh
          - --host=database
          - --port=3306
          - --timeout=30
          - --strict
          - --
          - /usr/local/tomcat/bin/catalina.sh
          - run
```

Limraj, 2022-03-21:
@kinsamanka thank you for your suggestion, we would like to use the script proposed in the documentation: https://docs.docker.com/compose/startup-order/#:~:text=Compose%20always%20starts%20and%20stops,service%3A. ..% 22% 20
If you would like to take care of it, please visit: https://github.com/SCADA-LTS/Scada-LTS/issues/2137

Best wishes.

closed 2022-03-22 by Limraj