SCADA-LTS/Scada-LTS #2152: Can't activate more than 1 Bacnet IP datasource simultaneously

nduvernois, 2022-03-29:
I have about 130-140 BacNet IP datasources, but as soon as 1 is active, I can't activate the other, and neither can I discover new datapoints in the active datasource, or in any other. This is the same behaviour as in ScadaBR.

For example, after a Tomcat restart, I find this in the event list : 
8:57:11 : 'B519 sous-station': Initialization error: Address already in use: Cannot bind
Well, actually, there is no other datasource with the same IP address.

Steps to reproduce the behavior:
Datasources have been initially created in ScadaBR, and the MySQL databased has been dumped to disk and imported to MySQL 5.7
1. only one BacNet IP address is active
2. in datasources management, click on any other database to activate it

Configuration

OS: Windows Server 2016
Browser : Chrome
Releases: ScadaLTS 2.6.13, Apache Tomcat 9.0.59, MySQL 5.7.19
I also noticed the same behaviour with ScadaLTS 2.5.1.

Log from tomcat9-stdout:
INFO  2022-03-29T08:57:12,082 (com.serotonin.mango.rt.RuntimeManager.initializeDataSource:378) - Data source 'B519 sous-station' initialized 
I have not found anything relevant at the same timestamp in catalina, tomcat9-stderr or localhost logs.

Patrykb0802, 2024-07-12:
Hi @nduvernois,

Does the problem still occur? If so, would you be able to send logs and screenshots of your configuration?

Regards,
@Patrykb0802

nduvernois, 2024-10-22:
so, it was just my fault by lack of knowledge in network configuration. The issue was caused by indicating port 47808 on the datapoint configuration for multiple Bacnet sources. May I suggest to add in the comment of bacnet datasources that this "port" field should remain at 0 unless the user knows well enough what he is doing ?

Limraj, 2024-10-22:
Hi @nduvernois,
1. Was the problem that more than one Data Source Bacnet had the same port set? 
2. Is it that multiple points were used in one Data Source with the port set to a value other than 0?

Regards,
Kamil Jarmusik

nduvernois, 2024-10-22:
Proposal 1  : more than one Bacnet Data Source had the same port set (= the port number in BacNet properties block).

Limraj, 2024-10-22:
Thanks, I think we can add such validation so that the Data Source cannot be saved with a busy port.

nduvernois, 2024-10-22:
Maybe it would also be worth precising in the BACnet I/P properties block what is relevant for the distant device (Device ID + broadcast address) and what is relevant for the ScadaLTS server (port, if I correctly understood)

nduvernois, 2024-10-25:
Last comment : when a new Bacnet datasource is created, by default, the field "port" is filled with 47808, whereas I had to input 0 in order to solve my issue. This is ... misleading.

Patrykb0802, 2024-11-07:
Edit:

Port validation will be added so now you won't be able to save data source with already taken port as well as some additional information about ports in help button on data source edit screen

closed 2025-04-11 by Limraj