SCADA-LTS/Scada-LTS #2159: Wrong status on new cmp components

grzesiekb, 2022-04-04:
**Describe the bug**
Wrong status on new cmp components

**To Reproduce**
Upload the project from the attachment and check whether the statuses are the same on the old and the new component
**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**

Limraj, 2022-04-04:
[image]
[image]
[image]
[image]
[image]
[image]

Limraj, 2022-05-12:
@grzesiekb Can we close this issue?

closed 2022-05-18 by Limraj