SCADA-LTS/Scada-LTS #2162: Avoid double input in data point properties edition

nduvernois, 2022-04-05:
There are quite a few things indicated in data point properties, some of them are more or less duplicate.
Example 1 : there is the "scientific unit" field, but also, if we select analog formatting, the suffix which naturally contains the unit. By the way, what is the current actual use of the "scientific unit" field ?
Example 2 : the tolerance used for data recording, when used, will most probably match the analog format : if one choses to display a number as #,###.#, there is a high probability that 0.1 would be a meaningful tolerance for data storage.

1) if a scientific unit is selected, then the analog format should be automatically pre-selected, and the suffix should be filled with the appropriate short version of the unit
2) once an number format is chosen (#.# for example), then the tolerance should be filled with the matching precision (so 0.1 in this case)
3) the block "text renderer properties" should be placed between the "point property" and the "logging properties" blocks. It would then also make sense to keep "chart rendered properties" close to "text renderer properties".
4) why is the "purge after X years" not included in the "system settings" ? Or at least a general default value should be part of system settings, and could be modified case by case for each datapoint ?

nduvernois, 2022-04-15:
Additional idea : we could also imagine a way to configure that, by default, a measurement point in volts is stored X years with a precision of 0.1 and displayed as #.# with suffix V, and so on for each any unit. This would save a lot of time during the configuration of new datapoints.

Limraj, 2025-03-07:
Hi @nduvernois,
Ad. 1. Ad. 2 Ad. 3 - We will implement this points;
Ad. 4 part It is available for setting from System Settings (Data Retention):
[image]
However, there is actually no global parameter for this functionality for the point value. We will create a new issue for this.

The problem with default tolerance is that this tolerance is closely related to whether the next value will be saved in the database or not, if the difference between the old and new value is less than the tolerance then the value will not be saved in the database, which is something completely different than the format of displaying this value, which comes down to cutting out the numbers, has no effects in the database. Therefore, we would prefer it to be 0 by default, which means that it saves the value in the database regardless of how much it has changed, so that the user himself consciously sets this parameter as he likes.

closed 2025-03-07 by Limraj