SCADA-LTS/Scada-LTS #2195: Addition of additional information in the company heading

gb-softq, 2022-05-05:
[image]

fabiodurao, 2022-05-05:
[image]

I would like to understand the central idea and proposal for the system. It seems to me that they plan to leave the system header as the final project header, let me present another point of view.

-----------------

In my previous projects I always created a layout in inkscape that already contained an area to be used as a header and another to be used as a space for the dynamic components all in a 1920x1050 image (30px less because of the footer copyright to not have scrollbar in fullscreen), in this header I usually put the links/navigation menu between the graphic screens, current time, logged in user (using a server side script) and even the html button for logout calling "logout.htm" in addition to some extra functions as in the example of the image, the temperature and local humidity captured from an API of HG Weather and finally in the upper left part was always placed the logo of the company that hired the project and then the title of the pages.

For this to work, I usually leave the final project in fullscreen and limit the pages that the final client will access (the client only browses the pages of the graphic representation), this restricts him from changing datasources, which is no longer a practice, but it also restricts him from accessing datapoint details, graphs and statistics that exist there. The alarm list part was not a problem, because a screen was created with only the alarm list component, although in this case it does not have alarm query tools as it exists by default in the system's alarm screen. A recurring demand was to have the reports screen accessible to the customer, this was made possible using an html component that called the reports screen within a graphical representation, however, hiding the menu. Follow:

```
 <script>

function hideHeaders(iframe){
    var contents = jQuery(iframe).contents();
    contents.find("#mainHeader").hide();
    contents.find("#subHeader").hide();
}

</script>
<iframe onload="hideHeaders(this)" width="300" height="300"  src="http://localhost:8080/ScadaBR/reports.shtm"></iframe>
```

Of course, this was a lot of work and it still limited the end user a little, but it served well.

Obs.: The navigation header I used to leave to create for last, because then I exported the json, copied the code of the last components that were referring to the header and imported the same code in the other pages keeping an equal header for all pages, only changing what was necessary. Ex.: Changing the color of a button to show that it was already accessing that screen.

---------------------

I would really like to understand what are the plans for future navigation, integrating everything, having a header area is great, however, it restricts the developer's options in system customization and in issues of color harmonization, themes and web designer. I think a long term path is to have individual components that can be customized and used no matter what layout the user uses. Eg: Having a ready component with an icon that shows the user logged in with submenus to edit the profile and logout that can be positioned wherever the developer wants and that can have a customized theme.

grzesiekb, 2022-05-10:
It just occurred to me that the header could be treated the same way as the graphical views and everyone could set it as they wish.

SoftQ, 2022-05-13:
Hello Fabio
It's a great idea to create a menu/navigation component that would be
easily customizable in terms of styling.  We shall carefully design such
component on a generic level- can You share some example/empty project with
customized header? It's ok to send it via email etc.

Best Regards
Michał

fabiodurao, 2022-05-13:
Sure, which email can I send it to?

Limraj, 2022-05-13:
I think you can here: scadalts-support@softq.pl and kamil.jarmusik@softq.pl

fabiodurao, 2022-05-20:
Okay, I just sent you a video by email demonstrating how headers are created, I don't have any projects ready here, I had to remotely access a client's server (I asked for permission to do so) and recorded it.

closed 2022-10-01 by Limraj