SCADA-LTS/Scada-LTS #2259: Fixed getCacheEntry failed: Cache service responded with 503

Limraj, 2022-06-13:
**Description**
There was a problem with the actions/cache@v2 

**Screenshots**
[image]
[image]

Limraj, 2022-06-13:
up version to v3 not work

RWilliams-MicronAgritech, 2022-06-13:
> up version to v3 not work [image]
> 
> edit: up version: *actions/checkout from v2 to v3 *actions/setup-node from v1 to v3
> 
> same problem

I use checkout v2, and my actions are now running.

`    steps:
    - uses: actions/checkout@v2
      name: Use Node.js ${{ matrix.node-version }}
    - uses: actions/setup-node@v2
      with:
        node-version: ${{ matrix.node-version }}
        # cache: 'npm'`

closed 2022-06-22 by Limraj