SCADA-LTS/Scada-LTS #2263: Following steps on https://github.com/SCADA-LTS/Scada-LTS/wiki/ gives me Tomcat 404 error

Kortenbach, 2022-06-15:
**Describe the bug**
After following the steps in https://github.com/SCADA-LTS/Scada-LTS/wiki/ for creating a docker container I get a Tomcat 404 error.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '(https://github.com/SCADA-LTS/Scada-LTS/wiki/)'
2. Folow the two step startup (docker-compose up database. And then: docker-compose up scadalts) and confirm there are no major errors.
3. Verify the containers are running (I used docker desktop in Win10)
4. Open a browser and type "http://localhost:8080/" or "http://localhost:8080/ScadaLTS"
5. Watch the error message: "HTTP Status 404 – Not Found", "Type Status Report", "Description The origin server did not find a current representation for the target resource or is not willing to disclose that one exists."

**Expected behavior**
I would have expected to see the opening screen of Scada-LTS

**Desktop (please complete the following information):**
 - OS: Windows 10 + Docker desktop
 - Browser Chrome

**Additional context**
This is my first try to work with Scada-LTS. I know only a little bit about Docker / Linux. I'm mainly a PLC programmer looking for an open source SCADA. I've checked many resources for this issue but found none.

Limraj, 2022-06-15:
Hello,
try:
http://localhost:8080/Scada-LTS/

Best wishes.

Kortenbach, 2022-06-15:
Hi Limraj,

Thank you for taking the time to answer my question. 
I've tried many variants like \ScadaBR\, \scada-LTS\, \Index.html\ but this one did the trick! 
It may be a good idea to add this to the quick-start part of readme.md cuz it wasn't obvious (at least for me!).

Best regards.

Kortenbach, 2022-06-15:
P.S. the MASTER image was not available, so I changed it to "scadalts/scadalts:latest" in the docker-compose.yml file

Limraj, 2022-06-15:
Thanks for your comments, I have introduced some of them to the wiki. ("login / pass" of the default user, information about the latest tag.)