SCADA-LTS/Scada-LTS #2264: Switching between old / new user interface not consistent / clear

Kortenbach, 2022-06-15:
**Describe the bug**
When clicking "alarms" in Scada-LTS V2.6.17. The user interface changes to a fresh new "green" theme. In this new UI only alarms related actions can be selcted. The bottom action "old UI" takes me back to the "old" ui.

**To Reproduce**
Steps to reproduce the behavior:
1. Install Scada-LTS V.2.6.17 (maybe others have this problem as well)
2. Click on the "alarms" icon (btw, the icon that is left from "alarms" is also called "alarms". I think it should be called "events(?)")
3. Watch the change to the new "green" UI theme. (btw, the button says "acknowledge störung/alarms" störung is a German word in an otherwise english UI)
4. The only way to get all menu items back is to go back to the old UI by clicking "Old UI" icon.

**Expected behavior**
I would expect either a consistent old UI or a consistent new UI and a way to switch between them. Now I have half the application running the old- and half the application running the new UI and I don't know how to switch between them.

**Desktop (please complete the following information):**
 - OS: Win 10 + Docker
 - Browser: Chrome

**Additional context**
I am a first time user of SCADA-LTS.

Limraj, 2022-06-15:
The new interface is under development, it is under the "Alarms" button (there are two "Alarms" buttons, one for the old alarms and the other for the new ui). 
However, from version 2.7.1 it is more intuitively solved, the new interface has a new button.

Best wishes.

closed 2022-06-16 by Kortenbach

Kortenbach, 2022-06-17:
Thanks for your comment!