SCADA-LTS/Scada-LTS #2265: Modbus TCP errors at refresh-rate < 2000ms

Kortenbach, 2022-06-16:
**Describe the bug**
Using tool "ModbusRsSim2" I found that Modbus TCP communication had problems when update-period is <2s.

**To Reproduce**
Steps to reproduce the behavior:
1. Create a datasource with an update period of 1s pionting to your PC's own ip address
2. Add a  few datapoints to this datasource (I added 3 coils)
3. Activate points and datasource
4. Download and start ModbusRsSim. Select Modbus TCP/IP from the right drop-down box.
5. Once communication is established press the "Comms" button to view the comms details in ModbusRsSim
6. The communication works but stutters. 

**Expected behavior**
I expected stutter free comms (using 2000ms update period). Trace:
```
16:31:34.941 [7636] Thread listening for connection..
16:31:35.889 (840)[13364] (Socket)[Thread] Connection accepted.
16:31:35.889 New Connection from IP Address:Port = 192.168.2.5:502.
16:31:35.894 RX:01 7A 00 00 00 06 01 01 00 00 00 03 
16:31:35.918 TX:01 7A 00 00 00 04 01 01 01 05 
16:31:35.968 (840)[13364] (Socket)[Thread] Connection Closed. Closing socket normally.
16:31:36.934 [13364] Thread listening for connection..
16:31:37.889 (840)[4212] (Socket)[Thread] Connection accepted.
16:31:37.889 New Connection from IP Address:Port = 192.168.2.5:502.
16:31:37.902 RX:01 7B 00 00 00 06 01 01 00 00 00 03 
16:31:37.918 TX:01 7B 00 00 00 04 01 01 01 05 
16:31:37.968 (840)[4212] (Socket)[Thread] Connection Closed. Closing socket normally.
16:31:38.933 [4212] Thread listening for connection..
```

But at 1000ms update period I get errors in Scada-LTS and this trace:
16:41:15.570 [9188] Thread listening for connection..
16:41:15.597 Read Output Coils from 0 for 3 bits.
16:41:15.597 Write error on Listen socket : Socket operation on non socket
16:41:15.597 [5724] Thread listening for connection..
16:41:19.562 (840)[7636] (Socket)[Thread] Connection accepted.
16:41:19.562 New Connection from IP Address:Port = 192.168.2.5:502.
16:41:19.571 RX:00 CE 00 00 00 06 01 01 00 00 00 03 
16:41:19.590 TX:00 CE 00 00 00 04 01 01 01 05 
16:41:19.641 (840)[7636] (Socket)[Thread] Connection Closed. Closing socket normally.

**Desktop (please complete the following information):**
 - OS: Win 10
 - Browser: Chrome

**Additional context**
When testing this with an other Modbus tool (QModbusMaster) a refreshrate of 1000ms doesn't give me any problems. When looking at the two traces I noticed that Scada-LTS closes the TCP connection after each (read) operation. QModbusMaster doesn't do this. It seems inefficient to do so and might be causing the communication problems. I havent tested this with another Modbus slave simulator so it could be that ModbusRsSim2 is just sensitive when it comes to repeatedly opening and closing the connection. Anyway, I thought I'd share this...

Best regards.

Limraj, 2022-06-16:
Thanks we will look at this case.

We use our ModbusPal fork:
https://github.com/SCADA-LTS/ModbusPal

Kortenbach, 2022-06-16:
Hi Limraj,
Your quick responses are highly appreciated!
Here your go:
`{
   "dataSources":[
      {
         "xid":"DS_547698",
         "type":"MODBUS_IP",
         "alarmLevels":{
            "POINT_WRITE_EXCEPTION":"URGENT",
            "DATA_SOURCE_EXCEPTION":"URGENT",
            "POINT_READ_EXCEPTION":"URGENT"
         },
         "updatePeriodType":"SECONDS",
         "transportType":"TCP",
         "contiguousBatches":false,
         "createSlaveMonitorPoints":false,
         "createSocketMonitorPoint":false,
         "enabled":true,
         "encapsulated":false,
         "host":"192.168.2.5",
         "maxReadBitCount":2000,
         "maxReadRegisterCount":125,
         "maxWriteRegisterCount":120,
         "name":"local modbus sim",
         "port":502,
         "quantize":false,
         "retries":2,
         "timeout":1000,
         "updatePeriods":1
      }
   ],
   "dataPoints":[
      {
         "xid":"DP_254024",
         "loggingType":"ON_CHANGE",
         "intervalLoggingPeriodType":"MINUTES",
         "intervalLoggingType":"INSTANT",
         "purgeType":"YEARS",
         "pointLocator":{
            "range":"COIL_STATUS",
            "modbusDataType":"BINARY",
            "additive":0.0,
            "bit":0,
            "charset":"ASCII",
            "multiplier":1.0,
            "offset":0,
            "registerCount":0,
            "settableOverride":true,
            "slaveId":1,
            "slaveMonitor":false,
            "socketMonitor":false
         },
         "eventDetectors":[
         ],
         "engineeringUnits":"",
         "chartColour":null,
         "chartRenderer":null,
         "dataSourceXid":"DS_547698",
         "defaultCacheSize":1,
         "description":null,
         "deviceName":"local modbus sim",
         "discardExtremeValues":false,
         "discardHighLimit":1.7976931348623157E308,
         "discardLowLimit":-1.7976931348623157E308,
         "enabled":true,
         "eventTextRenderer":{
            "type":"EVENT_NONE"
         },
         "intervalLoggingPeriod":15,
         "name":"Coil0",
         "purgePeriod":1,
         "textRenderer":{
            "type":"PLAIN",
            "suffix":""
         },
         "tolerance":0.0
      },
      {
         "xid":"DP_286032",
         "loggingType":"ON_CHANGE",
         "intervalLoggingPeriodType":"MINUTES",
         "intervalLoggingType":"INSTANT",
         "purgeType":"YEARS",
         "pointLocator":{
            "range":"COIL_STATUS",
            "modbusDataType":"BINARY",
            "additive":0.0,
            "bit":0,
            "charset":"ASCII",
            "multiplier":1.0,
            "offset":1,
            "registerCount":0,
            "settableOverride":true,
            "slaveId":1,
            "slaveMonitor":false,
            "socketMonitor":false
         },
         "eventDetectors":[
         ],
         "engineeringUnits":"",
         "chartColour":null,
         "chartRenderer":null,
         "dataSourceXid":"DS_547698",
         "defaultCacheSize":1,
         "description":null,
         "deviceName":"local modbus sim",
         "discardExtremeValues":false,
         "discardHighLimit":1.7976931348623157E308,
         "discardLowLimit":-1.7976931348623157E308,
         "enabled":true,
         "eventTextRenderer":{
            "type":"EVENT_NONE"
         },
         "intervalLoggingPeriod":15,
         "name":"Coil1",
         "purgePeriod":1,
         "textRenderer":{
            "type":"PLAIN",
            "suffix":""
         },
         "tolerance":0.0
      },
      {
         "xid":"DP_564240",
         "loggingType":"ON_CHANGE",
         "intervalLoggingPeriodType":"MINUTES",
         "intervalLoggingType":"INSTANT",
         "purgeType":"YEARS",
         "pointLocator":{
            "range":"COIL_STATUS",
            "modbusDataType":"BINARY",
            "additive":0.0,
            "bit":0,
            "charset":"ASCII",
            "multiplier":1.0,
            "offset":2,
            "registerCount":0,
            "settableOverride":true,
            "slaveId":1,
            "slaveMonitor":false,
            "socketMonitor":false
         },
         "eventDetectors":[
         ],
         "engineeringUnits":"",
         "chartColour":null,
         "chartRenderer":null,
         "dataSourceXid":"DS_547698",
         "defaultCacheSize":1,
         "description":null,
         "deviceName":"local modbus sim",
         "discardExtremeValues":false,
         "discardHighLimit":1.7976931348623157E308,
         "discardLowLimit":-1.7976931348623157E308,
         "enabled":true,
         "eventTextRenderer":{
            "type":"EVENT_NONE"
         },
         "intervalLoggingPeriod":15,
         "name":"coil2",
         "purgePeriod":1,
         "textRenderer":{
            "type":"PLAIN",
            "suffix":""
         },
         "tolerance":0.0
      }
   ]
}`

Limraj, 2022-06-16:
Thanks. When I have a moment, I'll take a look at it.
Remember that you can attach files, you need to click on "Attach files ..." at the bottom of the comment window.

Best wishes.

Kortenbach, 2022-06-16:
Thank you Limraj.
Have a nice evening.

fabiodurao, 2022-06-17:
I imported your json and in a quick check I verified that it is using the update every 1 second (1000ms) and the timeout too, but it still allows 2 more retries, in this case the total time until it generates an alarm in case of failure, is 3000ms, try to put a maximum of 300ms timeout or 450ms and decrease it to 1 retry. Here's an example:
[image]

fabiodurao, 2022-06-17:
Another change that can help you not to close the socket is to change from "TCP" to "TCP keep alive".

Kortenbach, 2022-06-17:
Hi Fabio,
Thank you for looking into this issue.
The TCP keep alive option sounds promising.
I will try next week.

Kortenbach, 2022-06-18:
That's what I understood from the last message. Will try asap.

Limraj, 2022-06-22:
Hi, has the problem been eliminated?

Kortenbach, 2022-06-22:
Hi Limraj,
I just tried the "TCP with keep alive" option.
Now the connection is not terminated after each communication.
Refreshrates <200ms are working flawlessly.
Thanks again for you help and happy coding!

closed 2022-06-22 by Kortenbach