SCADA-LTS/Scada-LTS #2268: ERROR in docker-compose up with docker-compose.yml

rvbatista, 2022-06-23:
**Describe the bug**
got this error after docker-compose up with docker-compose.yml from https://github.com/SCADA-LTS/Scada-LTS/wiki
ERROR: manifest for scadalts/scadalts:master not found: manifest unknown: manifest unknown

**To Reproduce**
Steps to reproduce the behavior:
1. download docker-compose.yml from https://github.com/SCADA-LTS/Scada-LTS/wiki
2. run docker-compose up
3. wait.
4. See error

**Expected behavior**
containers running.

**Desktop (please complete the following information):**
 - OS: ubuntu 18.04

**Additional context**
running in proxmox.

rvbatista, 2022-06-23:
Had to modify docker-compose.yml from:
        image: scadalts/scadalts:master
to:
        image: scadalts/scadalts:v2.6.17

Limraj, 2022-06-24:
Hello,
I corrected the file from the wiki, you can use the tag latest. I would be grateful for confirmation if it is okay.

Best wishes.

closed 2022-06-24 by rvbatista