SCADA-LTS/Scada-LTS #2310: gradle build war lost node_modules folder

yanggjjie998, 2022-08-03:
*Expected Behavior*
I want to be able to build a complete war file.

*Actual Behavior*
use gradle build war task, Scada-LTS.war build success, but resources folder lost node_modules folder.
gradle build buildRun  is also lost node_modules folder.

*Explain*
Only changed SCADA-LTS/build.gradle tomcatVersion 9.0.53 to 9.0.41.
Am I missing a configuration?

*Specifications*
Scada-LTS: branch v2.7.1
IntelliJ IDEA 2020.1.1 (Ultimate Edition)
node: v16.16.0
npm:v8.11.0
gradle:v6.8.1

*image*
[image]

Limraj, 2022-08-03:
Hello,
You need to execute task: installWebDependency
You can also use the link that connects build, installWebDependency, deploying and running tomcat in debug mode: buildRunDebugDev. Task buildRun, or "build war" was not executing installWebDependency due to build time. 
Before doing these tasks, you had to have previously created node_modules, from a different build process.

Best wishes.

yanggjjie998, 2022-08-03:
> Hello, You need to execute task: installWebDependency You can also use the link that connects installWebDependency, build, war, deploying and running tomcat in debug mode: buildRunDebugDev. Task buildRun, or "build war" was not executing installWebDependency due to build time. Before doing these tasks, you had to have previously created node_modules, from a different build process.
> 
> Best wishes.

Thank you for replying to my question so quickly. According to the information you provided, I solved the problem.

closed 2022-08-03 by yanggjjie998

Limraj, 2022-08-03:
Also thanks for the detailed description and quick reply.