SCADA-LTS/Scada-LTS #2316: Customization of ScadaLTS

odenisnobre, 2022-08-28:
How would it be possible to customize ScadaLTS by changing data such as logos, colors when it is running in a docker container?
How to map the ScadaBR folder out of the container?

Limraj, 2022-08-28:
From no stable v2.7.1.1 docker image: scadalts/scadalts:pr-2308;

1) **Change styles global.** (**editor only works with firefox, from 2.7.1**) We can change the appearance of ui by setting custom csses in: 'System settings -> Custom stylesheet' past:
```
#viewContainer .viewComponent div div {
border-bottom-right-radius: 25px;
background-color: blue;
font-family: \"Copperplate\", Copperplate, fantasy;
border: 2px solid grey;
font-weight: bold;
font-size: 15px;
color:yellow;
}

#viewContainer .viewComponent input { 
border: 2px solid grey;
border-bottom-right-radius: 25px;
}

#viewContainer .viewComponent a {  
color:black 
}

#viewContainer .viewComponent .infoData {  
color:yellow
}

.hgl-editor, .hgl-highlighting {
height:400px
}
```
[image]

2) **Change styles for user.** (in 2.6.17) We can change the theme / appearance for a given user (Users-> specific user -> Theme), you can create a dedicated theme for you, but it would require our commitment;

3) **Change styles for a specific graphical view.** (in 2.6.17) For You can also add an HTML component on the view and set styles in it, I create an example configuration, where I changed the appearance of the popup to change the value. To edit an HTML component:
```
<style>
.viewComponent div div {
  border-bottom-right-radius: 25px;
  background-color: orange;
  font-family: \"Copperplate\", Copperplate, fantasy;
  border: 2px solid red;
  font-weight: bold;
  font-size: 15px;
  color:yellow
}
.viewComponent input {
  border: 2px solid red;
  border-bottom-right-radius: 25px;
}
.viewComponent a {  
 color:black
}
.viewComponent .infoData {  
 color:yellow
}
</style>
```
[image]

4) **Change Logo** (in 2.7.1) - We don't have a function that the user can use, but you can get this effect like this:
I. Upload image with logo: Graphical Views -> New view -> Choose file -> Upload images, and click right on graphic and click 'Open image in new tab', copy url as _url_image_, go back view, no save;
II. Go to 'System settings -> Custom stylesheet' past, and set _url_image_: 
```
#logo {
display:block;
-moz-box-sizing: border-box;
box-sizing: border-box;
background: url(url_image) no-repeat;
width:172px;
height:32px;
padding-left:172px;
}

.hgl-editor, .hgl-highlighting {
height:400px
}
```

odenisnobre, 2022-08-28:
Hi, wow, excellent answer.

How would I change the ScadaLTS logo?

Limraj, 2022-08-28:
It is at point 4.

odenisnobre, 2022-08-28:
Oh ok.
Sorry for the lack of attention.

Thank you very much.

Limraj, 2022-08-28:
Okay, if you have any more problems write to scadalts-support@softq.pl, kamil.jarmusik@softq.pl.

Best wishes.

Limraj, 2022-08-28:
@odenisnobre Can we close the topic?

closed 2022-08-28 by odenisnobre