SCADA-LTS/Scada-LTS #2346: Development enviroment

Xt-Man, 2022-10-03:
Hi, i'm following this link [https://github.com/SCADA-LTS/Scada-LTS/wiki/Getting-started---Development](https://github.com/SCADA-LTS/Scada-LTS/wiki/Getting-started---Development), but when running **gradle buildRunDebugDev** i'm getting
`C:\dev\Scada-LTS>gradle buildRunDebugDev
> Task :Scada-LTS:installWebDependency FAILED
> Task :buildRunDebugDev FAILED

FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':Scada-LTS:installWebDependency'.
> A problem occurred starting process 'command 'npm''

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output. Run with --scan to get full insights.

* Get more help at https://help.gradle.org

BUILD FAILED in 2s
1 actionable task: 1 executed`
what i'm forgotting?
thanks

Limraj, 2022-10-03:
What version of Gradle and OS do you have?

Xt-Man, 2022-10-03:
I'm using Win10, Gradle 6.8.1 on branch develop, i'll try last release branch

Xt-Man, 2022-10-03:
I'm not interested in the new vue.js front, can i ignore it?

Limraj, 2022-10-03:
Paste this file in /Scada-LTS/scadalts-ui/ :
[package-lock.json.zip](https://github.com/SCADA-LTS/Scada-LTS/files/9698811/package-lock.json.zip)

And this file in /Scada-LTS/
[build.gradle.zip](https://github.com/SCADA-LTS/Scada-LTS/files/9698810/build.gradle.zip)

There are fixes for windows in version 2.7.2, so you can also link to branch release/2.7.2;
In a few days the patches will be merged into develop.

Limraj, 2022-10-03:
How does it work? If there is still a problem, write what happens next.

Xt-Man, 2022-10-04:
Hi @Limraj , it's working!
I really appreciate your help
Thanks!

Limraj, 2022-10-17:
Hi @Xt-Man ,
I tried to reproduce it locally and from the docker and failed to do it.
I mean, I don't have any views, I'm trying to add a new view and I don't see any errors, it looks like it's okay, you can save the view.

Have you tried to do what I wrote about? It is okay? 
Admittedly, the configuration would allow the conditions to be reproduce case the problem was related to a characteristic configuration, but maybe not be necessary.

If the problem persists, I would need more detailed information, e.g. what did you try to add to the view? The best detailed steps are what you clicked on and when it appeared.

Xt-Man, 2022-10-17:
Hi @Limraj ,
Yes, i tried what you wrote, it´s working now
Thank you

closed 2022-10-17 by Xt-Man