SCADA-LTS/Scada-LTS #2367: Data Source SQL to PostgreSQL problem

JRRedigolo, 2022-10-17:
We are using version 1.1. In our application we use SQL datasources to communicate with a Postgresql database. It happens that ScadaBR opens a connection with the bank in the first query and does not close it anymore. This causes a serious problem in the database that limits the number of open connections (default : 100 connections). When this number is reached, the bank starts to ignore the next queries, causing errors in the data visualization. We ask if it is not possible to add a mechanism to close the connection as soon as the query is performed, reopening it in the next query?
Thank you very much!

Limraj, 2022-10-19:
Hello,
we will check it and try to fix it in version 2.7.3.

Limraj, 2022-11-15:
Hi @JRRedigolo ,
we are dealing with this problem right now.
Could you please post an example configuration of one such Data Source with points? Of course without real ports/user/passwords etc.

This data source can be used in various ways, it is worth checking the specific scenario.

To generate this configuration:
1. Go to "Import/Export";
2. In the "Export" section, check the following checkboxes: Data Sources and Data Points;
3. Click on the "Generate JSON" button;
4. Copy json to notepad, edit text, leave specific Data Source and Data Point, change sensitive data and save;
5. Zip the text file with the configuration and attach it here.

JRRedigolo, 2022-11-16:
> Enviada em: terça-feira, 15 de novembro de 2022 13:58

Hello Kamil, good evening!

We are using version 1.1. We are now starting the migration to the latest stable version of the LTS, but the company we hired has a full schedule and still hasn't answered us. Would you like to receive the requested file even if it is version 1.1, or is it better to wait for the migration?

 

 

Atenciosamente,

 

 

José Roberto Redígolo

Diretor

*: 16 3618-5088  Cel: 16 98106-3737

* ***@***.***> ***@***.***

Nada é tão urgente que não possa ser feito com qualidade.

Esta mensagem contém informação confidencial e legalmente protegida aos seus destinatários. Se você não for uma destas pessoas, ou a tiver recebido por engano. fica desde já notificado de abster-se de divulgar, copiar, distribuir, examinar ou, de qualquer forma, utilizar a informação nela contida, por ser ilegal. Neste caso, solicita-se avisar imediatamente o remetente para as devidas providencias.

Limraj, 2022-11-16:
Hi,
I mean an example SQL Data Source configuration that you use and which causes problems, i.e. one such Data Source with at least one point, but you must remember not to send sensitive data, ports, etc.

Limraj, 2022-11-16:
4. If you use this production base in one application (ScadaBR 1.1) You can still consider configuring connections in the server itself. This means that the server itself closes unused connections. For example, in MySQL you can set it by changing the 'wait_timeout' parameter, the unit is seconds. There is also a 'max_connections' parameter in MySQL. Both of these parameters could stabilize the configurator on your code version. (ScadaBR 1.1)

Limraj, 2022-11-30:
Hello, the fix is ​​in version 2.7.3. I close the topic.

closed 2022-11-30 by Limraj

fabiodurao, 2023-03-20:
Hello,

I'm testing the JNDI resource a lot, in short I migrated the JRRedigolo server from ScadaBR 1.1 to Scada-LTS v2.7.4.1 and we have 85 SQL datasources, all are connecting to the same DB, they are just different SELECT's in each DataSource, it has improved a lot the question of the number of connections, but there were still some questions.

1 - I based myself on the context.xml resource to replicate the connection to the JNDI, as explained, I would like to understand how I can manipulate the parameters to reduce the number of connections, from what I understand, the parameters that interest me would be
           maxActive="80"
           minIdle="10"
           initialSize="10"

from the following list

```
    <Resource name="jdbc/scadalts"
          auth="Container"
          type="javax.sql.DataSource"
          factory="org.apache.tomcat.jdbc.pool.DataSourceFactory"
          testWhileIdle="true"
          testOnBorrow="true"
          testOnReturn="false"
          validationQuery="SELECT 1"
          validationInterval="30000"
          timeBetweenEvictionRunsMillis="30000"
          maxActive="80"
          minIdle="10"
          maxWait="10000"
          initialSize="10"
          removeAbandonedTimeout="1000"
          removeAbandoned="true"
          abandonWhenPercentageFull="75"
          logAbandoned="true"
          minEvictableIdleTimeMillis="30000"
          jmxEnabled="true"
          jdbcInterceptors="org.apache.tomcat.jdbc.pool.interceptor.ConnectionState;
                                                        org.apache.tomcat.jdbc.pool.interceptor.StatementFinalizer;
                                                        org.apache.tomcat.jdbc.pool.interceptor.ResetAbandonedTimer;
                                                        org.apache.tomcat.jdbc.pool.interceptor.SlowQueryReport(threshold=1500)"
          username="root"
          password="root"
          driverClassName="com.mysql.jdbc.Driver"
          defaultTransactionIsolation="READ_COMMITTED"
          connectionProperties="useSSL=false;allowPublicKeyRetrieval=true"
          url="jdbc:mysql://database:3306/scadalts"
         />
```
I don't know exactly how to use them, I did several tests and finally left them with minidle=10 and maxactive=10 and it didn't change much, it continues to open 40 connections. If you can explain what can be improved, I would appreciate it.

2 - There are several customers that will be migrated to Scada-LTS, in ScadaBR, all were as different applications within the same tomcat server and within the same instance in AWS, example:
IP:8080/client1
IP:8080/client2
IP:8080/client3
...
In this update I intend to use each client in a different instance, I just tested it that way, but I noticed that when I clone the instance, the new instance opens 40 more connections. We are connecting with the SQL datasource on an RDS service, there it is more robust and supports 400 connections, but we will easily reach this, since there are already 12 clients on the old server.

Hence the question, what is the best approach thinking about the issue of open connections? Everything on the same server would change something for being the same IP or create more connections per application in the same way.

3 - Finally, a print of the list of connected users follows, strange that it opens several connections with the query "SELECT1" as in the context.xml file, but opens some queries with the select from the datasource itself, but only a few, because they are 85 datasources in total.
[image]

Limraj, 2023-03-20:
Hi @fabiodurao,
1. The configuration you posted applies to mysql, not postgesql..
2. Unfortunately, there is something in tomcat that if you have something more than Scada-LTS/ScadaBR in webapps, it treats it as separate applications and applies the settings you enter in <context> separately to all applications, i.e. if outside of Scada-LTS you have e.g. docs and ROOT then you have to count these limits x3. The solution is to move the ROOT and the rest of the folders to another place, they are not needed for the operation of tomcat and Scada-LTS.

Limraj, 2023-03-21:
Hi @fabiodurao ,
what do you have in the CATALINA_HOME/webapps folder?

fabiodurao, 2023-03-21:
It only has the ROOT folder and the ScadaBR/Scada-LTS application. I can even remove the ROOT folder if you want me to test it, but it is no longer hitting the (n_applications)x10 account

[image]

Limraj, 2023-03-21:
1. ROOT you can move the directory higher so that it is only Scada-LTS.

2. Stop tomcat, see if all connections are terminated. (this may take a while)

3. I understand that you have one <Resources> defined with a field for postgresql?

4. When you do these things, take a screenshot with tomcat turned off, if there are any processes from the application, you need to close them, and then after starting the application, if you have changed the <Resource> configuration for postgresql, then also send a detailed configuration without ip address / logins / passwords.