SCADA-LTS/Scada-LTS #2398: Enable Private Vulnerability Reporting

g0dd42d, 2022-11-19:
**Describe the solution you'd like**
Enable GitHub's new **Private** Vulnerability Reporting feature.

Security > Private vulnerability reporting > Enable vulnerability reporting

Finally, update the security policy once this is deemed adequate.

**Additional context**
This would make it super easy to track, manage and fix any security-related issues without needing to email the developers. This would also fit the current process of tracking bugs/features/fixes. These reports would not be visible to the public until they are fixed and released by the maintainers.

Thanks!

SoftQ, 2022-11-19:
Hello,
Thank You for Your messages, please excuse the response delays. We will
take care of Your request next week.
First and foremost, Kamil has already implemented fixes and scheduled for
the next release that will be published soon.
Would You like to join our team?

Best Regards
Michał

Limraj, 2022-12-07:
Hi @wodzen,
Can you add private vulnerability issues now?

Best wishes,
Kamil Jarmusik

wodzen, 2022-12-09:
I can, thanks! I'll open issues as I test them on the most recent pre-releases.

wodzen, 2022-12-10:
Reopening to verify that you can now see the reports @Limraj. I'll add a couple more.

Limraj, 2022-12-27:
@wodzen 
I see the reports, so I close the topic.

closed 2022-12-27 by Limraj