SCADA-LTS/Scada-LTS #2405: SOAP error

Xt-Man, 2022-11-25:
**Describe the bug**
When i try do use SOAPUI tool (https://www.soapui.org/downloads/soapui/) i got an error
`Error loading [http://localhost:8089/Scada-LTS/services/API?wsdl]: org.apache.xmlbeans.XmlException: org.apache.xmlbeans.XmlException: error: The element type "img" must be terminated by the matching end-tag ""`
I´m running a ScadaLTS docker image

**To Reproduce**
Steps to reproduce the behavior:
1. Open SoapUI tool
2. Click on File -> New SOAP Project
3. Enter a Project Name (ScadaLTS)
4. Enter a wsdl url: http://localhost:8089/Scada-LTS/services/API?wsdl (in my case)

**Expected behavior**
The same steps on old ScadaBR works fine

i´m search in many files, web.xml, deploy.wsdd, api.wsdl, ScadaBRAPISkeleton.java, but i can´t find the problem, i appreciate any help

thanks

Limraj, 2022-11-25:
Hello, 
which version did you try? I don't remember any client using these services, although I came across it a few times in the code. 
Xml is ungrateful at debugging. I'm looking at implementations of this class ScadaBRAPISkeleton, a nightmare. What exactly do you need these services for?

Xt-Man, 2022-11-25:
Hi @Limraj ,
ScadaBRAPISkeleton is a really nightmare!
In production i have 2 softwares consuming the ScadaBR Soap. I want to migrate from BR to LTS. My first thinking was keeping the soap... 
I´ll change to rest. At the moment i discovered that soap getDataHistory is the same that /point_value/getValuesFromTimePeriod/xid/

Rest is the way to go

Thanks for help

Limraj, 2022-11-25:
The subject is interesting, why it stopped working, I think I'll be back next week with information if we take up the fix.

Limraj, 2022-11-28:
Hi @Xt-Man, 
I managed to import the wsdl, you need to point to the file in: /Scada-LTS/WebContent/wsdl/API.wsdl and not this endpoint. This endpoint is read only, you can run it in your browser (Http GET), it works.

This will allow you to create a project, then if you want to test some endpoint, you right click on the action and "New request" then you have to add cookies from Session ID to the headers after logging in, unfortunately this tool does not support the way we can log in to Scada, so you have to do it e.g. in the browser and then copy it from the cookie header.

[image]

Xt-Man, 2022-11-28:
Hi @Limraj,
I´ll check out and give a feedback, thanks a lot

Xt-Man, 2022-12-01:
Hi @Limraj ,
wich branch are you using?
I'm using release/2.7.2, fresh install machine
[image]
but it's not working here
thanks

Limraj, 2022-12-01:
Hi @Xt-Man,
You need to log in to Scada through the form in the browser, then after logging in to the website, right click "Inspect", then Network and look for a request, e.g. doLongPoll and there in the request header there will be Cookie, you need to transfer them to the tool, as you can see on my screen, go back to him.

In version 2.7.4 we want to add the ability to log in basic to make the integration more convenient. In this version there will also be a special user 'soap-services' with access to these services only.
I will describe the details after making these changes.

Best wishes,
Kamil Jarmusik

Limraj, 2022-12-16:
Hi @Xt-Man,
### 1. In version 2.7.4.1 there will be a security corrected for SOAP services:
1.1 Access to the endpoint from wsdl is available without logged:
/services/API?wsdl
1.2. Access to service operations requires Basic Auth, via Https.
1.3. A special soap-services user will be created with access only to SOAP services, you cannot login to the Application with it, user is disabled by default; (username: soap-services, default password: soap-services);

### 2. Integrate with SOAP services:

#### 2.1. You need to generate the keys:
1. Login to the user system on which you want to launch tomcat with the application, and execute the command there: (if keytool not installed go to JAVA_HOME/bin, keytool included with the JRE)

`keytool -genkey -alias tomcat -keyalg RSA -storepass your_password -keypass your_password -dname 'CN=Scada-LTS;O=Scada-LTS;OU=Scada-LTS'`

This command will create a file in home directory logged user:
.keystore

#### 2.2. Next, you need to update the tomcat configuration with https communication, for this purpose, in file:
`CATALINA_HOME/conf/server.xml `
 you need to add an additional connector in <Service> tag:

```
<Connector port="8443" protocol="org.apache.coyote.http11.Http11NioProtocol"
maxThreads="150" SSLEnabled="true" scheme="https" secure="true"
clientAuth="false" sslProtocol="TLS"
keystoreFile="${user.home}/.keystore" keystorePass="your_password"
compression="on"/>
```

#### 2.3. The third step is to enable the soap-services user, for this you need to log in to the Admin, go to the "Users" tab, find "soap-services" and uncheck the "disabled" box and then click "save".

### 3. Use SOAP services:
Address:
https://localhost:8443/ScadaBR/services/API?wsdl

The default password for soap-services user can be changed after logging in to the administrator, in the Users tab, find the soap-services user, enter the new password in the password field and click "save".

[image]

Xt-Man, 2022-12-16:
Hi @Limraj ,

great!

Thank you!

closed 2022-12-16 by Xt-Man