SCADA-LTS/Scada-LTS #2435: Container does not start after crash

fabiodurao, 2022-12-10:
Scenario: Scada-lts latest installed on ubuntu server 20.04 on AWS with 8GB of storage.

I didn't realize that the disk was filling up and this week it reached 100% disk and everything crashed, I increased the storage, expanded the disk and partition and the storage problem was solved, however, the application does not go up anymore, I would like help to solve this problem and see if I can recover the project, follow the feedback I had in the terminal.
[terminal_report.txt](https://github.com/SCADA-LTS/Scada-LTS/files/10200985/terminal_report.txt)

fabiodurao, 2022-12-10:
complementing, the installation was with docker-compose and the tag was latest, the version as far as I remember was 2.6.17

Limraj, 2022-12-10:
We do not recommend using docker in production. For the data to be permanently preserved, additional configuration of volumes is needed, but it is not the most stable solution. What was clogging up this disk?

Limraj, 2022-12-12:
Hello,
We do not recommend using docker in production, it requires additional volumes configuration, which, however, is not very stable:
https://docs.docker.com/storage/volumes/

Limraj, 2022-12-29:
@fabiodurao 
Can I close the topic?

fabiodurao, 2023-01-02:
Thanks for the feedback,

I couldn't recover the DB, but I had the backup json and I found it easier to upload a new container and import, because the history wasn't important. I think you can close the topic. The information that docker is not trusted is something new here on the project's github. The main way to distribute Scada-LTS is by docker and it is still the easiest and fastest way to install it, since we don't have an installer for windows or linux yet. Manual installation has become almost impossible in newer versions of linux, such as ubuntu 22.04, up to ubuntu 18.04, it is still possible to install mysql 5.7 from the repository, in newer ones, even using mysql-apt-config has not worked in some instances, I don't know if there are plans to update mysql to versions 8 onwards, I know that there is the intention of questdb, but from what I understand it is only for the pointvalues table.

I haven't experienced any instability with docker, the system has been running great for this project, I didn't know I would need additional volumes. The problem I had is more related to my decision not to take care of the storage size or the amount of history that would be saved. This problem of running out of storage space and having a problem with the database and not being able to recover more, I've seen it happen in installations of ScadaBR also without docker, I imagine it wouldn't be different in Scada-LTS without using docker.

Is there any future prospects to facilitate this process manually? We need to think more about the upgrade policy between versions of Scada-LTS. For example, upgrade from 2.6.18 -> 2.7.4 (not compatible with all database tables as a column was added in table xyz).

Limraj, 2023-01-02:
Hi @fabiodurao,
How could you lose all your data if you don't use docker, which requires volumes to be configured to persist data? This is due to the nature of docker, not our particular approach. Docker's use is tricky when you have to go beyond the standard configuration. Docker alone is practically not used in production, if it is combined with container orchestration systems such as Kubernetes. In our wiki you have instructions to install mysql 5.7 on newer ubuntu: [Quick install - Linux Ubuntu 20.04+](https://github.com/SCADA-LTS/Scada-LTS/wiki/Getting-started---Development)

closed 2023-01-02 by Limraj

Limraj, 2023-01-03:
@fabiodurao If you have indeed encountered any problems with Scada, it's best to create a separate issue for each such problem with a scenario that will enable you to reproduce it, or information that will enable it. This will enable us to fix the problem.