SCADA-LTS/Scada-LTS #2438: Report from WatchList doesn't work

Xt-Man, 2022-12-15:
**Describe the bug**
WatchList page (~/watch_list.shtm) has a button "Create report for this watch list", but doesn't work

**To Reproduce**
Steps to reproduce the behavior:
1. Go to watch_list.shtm page
2. Click on "Create report for this watch list"
3. The report page is called ~/reports.shtm?wlid=1, but datapoints from this wl are not included

**Expected behavior**
The datapoints from watch list must be included

**Screenshots**
[image]
[image]

Xt-Man, 2022-12-15:
Hi @Limraj ,

in file com/serotonin/mango/web/dwr/ReportsDwr.java, i change the method "createReportFromWatchlist" to:

    public ReportVO createReportFromWatchlist(int watchListId) {
        WatchListService watchListService = new WatchListService();
        WatchList watchList = watchListService.getWatchList(watchListId);
        if (watchList == null)
            return null;

        watchListService.populateWatchlistData(watchList);
        ReportVO report = new ReportVO();
        report.setName(LocalizableMessage.getMessage(getResourceBundle(), "common.copyPrefix", watchList.getName()));
        for (DataPointVO dp : watchList.getPointList()) {
            ReportPointVO rp = new ReportPointVO();
            rp.setPointId(dp.getId());
            rp.setColour(dp.getChartColour());
            rp.setConsolidatedChart(true);
            report.getPoints().add(rp);
        }

        return report;
    }

and solve the issue, if there is a better way to solve it, please told me
thank you

Limraj, 2022-12-16:
Hi @Xt-Man,
If you could take a look: 
https://github.com/SCADA-LTS/Scada-LTS/pull/2439

Xt-Man, 2022-12-16:
Hi @Limraj ,
That´s it!
Thank you

closed 2022-12-16 by Xt-Man