SCADA-LTS/Scada-LTS #2450: Determining the optimal environment installation path for the application

Limraj, 2023-01-02:
**Description.**
The goal is to discuss how to improve environment and application installation. An installer for linux has been prepared, but there may be problems with newer versions of Ubuntu. This should be verified and possibly corrected.

fabiodurao, 2023-01-02:
I don't know if those involved are originally available to help with this project, but I believe they can use the scadabr linux installer to make the scada-lts installer, this installer is even prepared for an ARM platform such as the Raspberry pi.
https://github.com/ScadaBR/ScadaBR_Installer

there is also an installer repository for windows using inno setup

https://github.com/ScadaBR/innacio

Limraj, 2023-01-02:
Looks like both solutions don't apply to database server installation... 
There is an installer in the link, you can test it.

fabiodurao, 2023-01-02:
Yes, they were still using the old default database, Derby.

Limraj, 2023-01-02:
You can try to configure such a database in tomcat and check how it works in terms of compatibility and performance:
https://db.apache.org/derby/integrate/DerbyTomcat5512JPetStor.html

Limraj, 2023-01-11:
@fabiodurao The point is that you try to run this installer and write what you see there, since it does not work. What errors specifically? There may be a problem with the new ubuntu here.

I was installing on the version given in the wiki ([Quick install - Linux Ubuntu 20.04+](https://github.com/SCADA-LTS/Scada-LTS/wiki/Getting-started---Development)). You write about some errors, you don't write what exactly. 

The following topics might be helpful:
1. What version of java do you have? 
2. You can see that you have root set everywhere in lib? Set there the user under which you run tomcat with Scada, all files and folders in tomcat should have this user set. (chown -R user:group dir) 
3. The logs show that the activator.jar file is broken. You need to download a new file and check md5 sum if it is downloading the correct one.

If your goal is something constructive, then approach the description of the problem in such a way that something comes out of it, something can be repeated, etc. 

Regards,
Kamil Jarmusik

fabiodurao, 2023-01-12:
I achieved!

1 - I was making a mistake and a lack of attention, the lack of attention is that in the context.xml, I did not change the database to localhost (why not leave the model file already with localhost, whoever uses an external DB will know that he needs to exchange). Now the silly error is that when I went to download the 4 files from the folder, I opened the github folder, I right-clicked on each file and copied the url, then I did a wget with this server link, result, download a file with the right name but with the html content from the github site as seen below.

2 - I had java right, I tried to do the md5sum, to check, in linux I gave the command "md5sum file name" and it calculated the hash, but I didn't find that number on github to consult and looking on the internet for it I didn't find it either, but soon doing this process I noticed the error mentioned above and attacked it, leaving the hash in the background.

3 - I followed this tutorial https://github.com/SCADA-LTS/Scada-LTS/wiki/Scada-LTS-Linux-installation-tutorial and installed mysql 5.7 with a tutorial like this link https://www. vultr.com/docs/how-to-install-mysql-5-7-on-ubuntu-20-04/ which is on the page you referred me to. I don't know why, but I've had problems installing mysql 5.7 on an ubuntu 20.04 even with the wget tip https://dev.mysql.com/get/mysql-apt-config_0.8.12-1_all.deb not opening in the list from options the option to make compatible with the version of ubuntu that has mysql 5.7 in the repository.

4 - About the installer, honestly I didn't even understand what needs to be done. What does the "inst" file do, I don't know what its extension is, I don't understand if I need to download all the files or just some, I don't understand if the installer already comes with all the installation files, if I need to download something else on my own or if it downloads itself during execution, also when I run /sh/start.sh the following happens

I achieved!

1 - I was making a mistake and a lack of attention, the lack of attention is that in the context.xml, I did not change the database to localhost (why not leave the model file already with localhost, whoever uses an external DB will know that he needs to exchange). Now the silly error is that when I went to download the 4 files from the folder, I opened the github folder, I right-clicked on each file and copied the url, then I did a wget with this server link, result, download a file with the right name but with the html content from the github site as seen below.

[image]

2 - I had java right, I tried to do the md5sum, to check, in linux I gave the command "md5sum file name" and it calculated the hash, but I didn't find that number on github to consult and looking on the internet for it I didn't find it either, but soon doing this process I noticed the error mentioned above and attacked it, leaving the hash in the background.

3 - I followed this tutorial https://github.com/SCADA-LTS/Scada-LTS/wiki/Scada-LTS-Linux-installation-tutorial and installed mysql 5.7 with a tutorial like this link https://www. vultr.com/docs/how-to-install-mysql-5-7-on-ubuntu-20-04/ which is on the page you referred me to. I don't know why, but I've had problems installing mysql 5.7 on an ubuntu 20.04 even with the wget tip https://dev.mysql.com/get/mysql-apt-config_0.8.12-1_all.deb not opening in the list from options the option to make compatible with the version of ubuntu that has mysql 5.7 in the repository.

4 - About the installer, honestly I didn't even understand what needs to be done. What does the "inst" file do, I don't know what its extension is, I don't understand if I need to download all the files or just some, I don't understand if the installer already comes with all the installation files, if I need to download something else on my own or if it downloads itself during execution, also when I run /sh/start.sh the following happens

```
ubuntu@ip-10-0-0-68:~/installer-0.5.1/sh$ sudo ./start.sh
/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/snap/bin:/home/ubuntu/installer-0.5.1/sh/jdk- 11.0.13+8/bin
/home/ubuntu/installer-0.5.1/sh/apache-tomcat-9.0.56
./start.sh: line 9: cd: ./apache-tomcat-9.0.56: No such file or directory
./start.sh: line 11: ./bin/catalina.sh: No such file or directory
tail: cannot open './logs/catalina.out' for reading: No such file or directory
tail: no files remaining
ubuntu@ip-10-0-0-68:~/installer-0.5.1/sh$
```

Still about the installation, one thing that bothers me in the default installation of tomcat by "sudo apt install tomcat..." is that it separates each part of tomcat in a different folder, (/usr/share/tomcat9/lib/ /etc /tomcat9/ /var/lib/tomcat9/webapps/), the ScadaBR installer put everything in the folder (/opt/ScadaBR), as I couldn't test the installer, I don't know if it's already like that, but it's something that makes it a lot easier management.

Thanks for all the support, I've really enjoyed Scada-LTS.

Limraj, 2023-01-13:
Hi @fabiodurao,

Ad. 2 You were just supposed to enter this hash here. However, if you already know how to download these things the right way, and you have found out that this jar was corrupted, then there is no need to do so. For future reference, if a file throws an error, it's a good idea to upload a hash md5 of that file.

Ad. 3 If you have an error, something doesn't work, then in order to be able to react somehow, you have to show it on screens, in logs in anything. Otherwise I can't guess what's going on what you've done. Put yourself in my place and think about what you would need to know to reproduce what you did or understand what is going on. If you can't install something from the manager, do it from the console.

Ad. 4 You need to run the inst file (I wrote you earlier that you need to run the inst file), give it permission to executable:
`chmod +x inst`
 and run
`./inst`
Anything you do that leads to an error should be thoroughly documented.

I'll tell you directly what's wrong with your cooperation. We did a lot of the things you said, and when I asked for tests, you disappeared and instead of going back to testing, you showed up with new topics. Then you write nonsense like I owe you MQTT when we've done a lot more stuff and the fact that you ignore it is just because of your negligence. Now the problems you describe really don't show much.

If you don't change your approach, I won't be able to help you. 

Regards,
Kamil Jarmusik

fabiodurao, 2023-01-13:
I'm sorry for the way I reported things, regarding the mysql installation error on another server, it was just an isolated comment, I no longer have access to the server to be able to show the logs, so understanding that I shouldn't have touched this point.

Regarding mqtt, I would like to have already been able to stop and test what was raised, but unfortunately I still haven't had the time, however, I will and reported it in the relevant issue.

I'm going to upload another ubuntu instance to test the installer and not conflict with the current installation, but it's still not clear which files will need to be downloaded, is it just the "inst" and the .tar.gz? Do they both need to be in the same folder? If it is possible to detail the procedure a little better, I would be grateful.

fabiodurao, 2023-01-17:
I created a new instance with ubuntu 22.04 in aws to test the installer.

I downloaded the "inst" file and the "installer-0.5.1.tar.gz" file, unzipped the .tar.gz file and made the "inst" executable.
When running the "inst" I got the following error:

```
ubuntu@ip-172-31-40-47:~$ ./inst
./inst: error while loading shared libraries: libssl.so.1.1: cannot open shared object file: No such file or directory
```

Searching, I managed to fix it by installing the lib with the commands:

```
ubuntu@ip-172-31-40-47:~/installer-0.5.1$ sudo wget http://nz2.archive.ubuntu.com/ubuntu/pool/main/o/openssl/libssl1.1_1.1.1f -1ubuntu2.16_amd64.deb

ubuntu@ip-172-31-40-47:~/installer-0.5.1$ sudo dpkg -i libssl1.1_1.1.1f-1ubuntu2.16_amd64.deb
```

Then I ran the installer again and got the following result:

```
ubuntu@ip-172-31-40-47:~$ ./inst
Start inst v0.5.1 for Scada-LTS v2.6.13
Internal Mysql on port: 9797
Internal tomcat on port: 8080
Internal java 1.11
Get java and unpacking

Get tomcat and unpacking
Get mysql and unpacking
thread 'main' panicked at 'called `Result::unwrap()` on an `Err` value: Custom { kind: InvalidInput, error: TarError { desc: "failed to iterate over archive", io: Custom { kind: InvalidInput , error: "invalid gzip header" } } }', src/inst.rs:265:64
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
ubuntu@ip-172-31-40-47:~$
```

I noticed that the installer already downloads and installs everything in the folder where it is, I believe that the most correct thing would be to run it inside the /opt folder

```
ubuntu@ip-172-31-40-47:~$ ls
apache-tomcat-9.0.56 inst installer-0.5.1 installer-0.5.1.tar.gz java.tar.gz jdk-11.0.13+8 my.tar.gz tomcat.tar.gz
```

I couldn't move forward from this step.

Limraj, 2023-01-17:
Okay, I got the same error, it looks like ubuntu switched to OpenSSL 3 and one of the libraries doesn't work, it would need to be updated in the installer. [ubuntu-22-04-libssl-so-1-1](https://stackoverflow.com/questions/72133316/ubuntu-22-04-libssl-so-1-1-cannot-open-shared-object-file-no-such-file-or-di)

I don't know if it makes sense to continue, you need to update the installer. However, if you want more things you can see there:

As for the next error, it looks like it is trying to unpack a file that has the wrong format, or one of the downloaded files is broken, you can download these files separately and check the sizes, preferably the hash md5. Here are links to these files:
https://github.com/SCADA-LTS/installer/blob/main/src/main.rs

But as I say, I'd rather not go on with this, perhaps the problem is also that you have two versions of openssl installed.

fabiodurao, 2023-03-06:
I tested the installer for windows and it works ok. I was able to run Scada-LTS, it went up to version 2.5. Since it downloads while running the script, would it be possible to download the latest version from github directly? If I download .war latest manually, stop tomcat, replace and start tomcat will it work or is there any restricted compatibility in this version?

Some suggested improvements:
- The files were all in the same inst.exe directory, in my case, in the downloads folder, it would be interesting on windows to install java in the c:/program files/java folder and tomcat also in the program files or ask the user;
- Maybe even do something like the traditional mysql installer does, call other installers, to give the option for the user to leave java as an environment variable for example;
- Apparently it doesn't stay as a service, but a prompt window is open, if you close it, tomcat stops, putting it as a service, with that old tomcat9w (with a graphical stop and start function) would be interesting;
- And all this running with a graphical interface, as already described in the items above, would be excellent, Inno Setup can help in this work. For Windows users, anything that can be easier is better.

I know it's still a 0.5.1 version, but I'd like to leave my suggestions for future implementations.

Regarding the installer for linux, I don't believe there are two versions of openssl, because I created a new instance on AWS with ubuntu 22.04 just to test the installer without possible defects, the only thing I did before that was to run a "sudo apt update" .

Limraj, 2023-03-06:
@fabiodurao thanks for the tests.

As for the installer, we'll try to adapt the installer you used for Scada-LTS so far. 

`Regarding the installer for linux, I don't believe there are two versions of openssl, because I created a new instance on AWS with ubuntu 22.04 just to test the installer without possible defects, the only thing I did before that was to run a "sudo apt update" .`
Please explain to me what are you writing about here, what exactly did you do and what are you talking about?

Limraj, 2023-03-07:
Here is the installer (Tomcat9 + Scada-LTS):
https://github.com/SCADA-LTS/windows-installer/releases/tag/v1.0.0

Limraj, 2023-03-15:
Hi @fabiodurao,
Update to version 2.7.5.1:
https://github.com/SCADA-LTS/windows-installer/releases/tag/v1.0.2

fabiodurao, 2023-03-25:
Hi Limraj,

Great news!
I tested the installer and it worked very well for me.
- I installed openjre 11 from this link https://www.openlogic.com/openjdk-downloads ;
- I installed mysql 5.7, created the database, user and password;
- I ran the installer and pointed to the folder where openjre11 was installed and ran the whole installer;
- Right after the installation it didn't work, because I hadn't made the link to the database;
- I stopped tomcat;
- I edited the file C:\Program Files\Scada-LTS\tomcat\conf\context.xml and passed the credentials for Scada-LTS to have access to the database;
- I started tomcat and it worked perfectly in the address
http://localhost:8080/Scada-LTS

Congratulations on your work!

Limraj, 2023-03-25:
Information about java, and install mysql server, is included here:
https://github.com/SCADA-LTS/Scada-LTS/wiki/Getting-started---Development

Limraj, 2023-06-07:
Installers have been created:
[windows-installer](https://github.com/SCADA-LTS/windows-installer/)
[linux-installer](https://github.com/SCADA-LTS/linux-installer/)

Hi @fabiodurao, Can we close the topic?

closed 2024-06-03 by Limraj