SCADA-LTS/Scada-LTS #2456: Error message on WatchList Get Chart

Xt-Man, 2023-01-04:
**Describe the bug**
An error message appears -> "Error marshalling int: Format error converting 419.77778. See the logs for more details."

**To Reproduce**
Steps to reproduce the behavior:
1. On yout browser, press ctrl- to reduce the zoom to 90%
2. On WatchList, click on Get Chart
3. See error
[image]

**Expected behavior**
Shows the graphic

**Additional context**
Any browser zoom (except 100%) show this error

Limraj, 2023-04-20:
Hello @Xt-Man !
would you like to improve it? We have to deal with it...

fabiodurao, 2023-04-20:
This error happened to me on slts 2.7.5.2 with chrome, but I was using zoom in the browser, when I went back to 100% I didn't have the problem anymore and the graph was built normally.

Xt-Man, 2023-04-20:
Hi @fabiodurao , exactly! The error message appears only on zoom != 100.

I will push a merge request with the fix

Xt-Man, 2023-04-20:
Hi, @Limraj @fabiodurao 
Done! Pull Request created
[https://github.com/SCADA-LTS/Scada-LTS/pull/2519]

Thanks

closed 2023-05-16 by Limraj