SCADA-LTS/Scada-LTS #2474: Unable to use .svg to Synoptic Panel

sheza160217, 2023-02-12:
Hi,

I have installed Scada-LTS on a virtual machine using docker.

_Installation: Docker on proxmox LXC container
Image: release-2.7.5
Method: docker-compose_

Issue: Followed your tutorial on Youtube for Synoptic Panels. When I try to upload a .svg fiel created in Inkscape to Synoptic Panel, there is a prompt to confirm adding the image. But there are no other details to assign any virtual point data to the svg components. The svg image is shown on the Synoptic Panel as a static image.

Thanks for your guidance.

Limraj, 2023-02-12:
Hi @sheza160217,
1. Which graphic did you try to upload?

2. In the video you have information that after uploading the image you need to additionally select this panel from the list, did you do this step?

Regards,
Kamil Jarmusik

Limraj, 2023-02-13:
[image]
I managed to add this graphic, as I said in point 2, after uploading the graphic you have to select the panel again in this select:
[image]

Limraj, 2023-02-13:
But there is an error, if you don't enter a name, then this panel is in this list, but without any inscription etc, it looks like it's not there. There should be a validation on the frontend side to force entering the name of the panel:

[image]

sheza160217, 2023-02-14:
@Limraj :

Sorry for taking time to reply. I have tried it again. This time with all the steps you have mentioend in above posts.

Still I do not get the complete diaglog to bind the data points. I have also reinstalled on a new LXC container with docker. I can confirm the behaviour is the same.

Is it casued by the docker image 2.7.5? Or something else? Not sure.

Thanks.

Limraj, 2023-02-14:
Hi @sheza160217,
I understand. This corrected svg: 
[image]

Okay, for this field to work, the id must match the regex:
`SLTS_((waterlevel|fan|point|state|valve)_\d|(_(background\b|background_right|background_left|value)))+\b`
So it is not enough that it starts with SLTS_, it must be followed by one of the 5 keywords (waterlevel, fan, point, state, valve), and at the end there must be a number, and it may or may not be: _background, _background_left, _background_right or _value.

Based on these ids, the component determines the function of the given field. Examples of valid ids:
SLTS_waterlevel_1
SLTS_waterlevel_1_background

SLTS_valve_1
SLTS_valve_1_background_left,
SLTS_valve_1_background_right,
SLTS_valve_1_value

SLTS_point_1

SLTS_fan_1
SLTS_fan_1_background
SLTS_fan_1_value

However, this is an experimental feature and from what I can see there are problems with it.

Limraj, 2023-02-16:
Hi @sheza160217,
The point_1/state_1 fields also need point_1_value/state_1_value. Then it works, the values ​​are refreshed and in the right place.

This corrected svg: 
[image]

Based on these ids, the component determines the function of the given field. Examples of valid ids:
(All are for point_1)
SLTS_point_1
SLTS_point_1_value

(All are for state_1)
SLTS_state_1
SLTS_state_1_value

Limraj, 2023-02-17:
Hi @sheza160217,
So it is important to fulfill this formula for ids of text fields for ids with: value, label, and polygons field for ids with: background, background_left, background_right and main field (examples: SLTS_point_2, SLTS_valve_3, SLTS_waterlevel_1). 

Additionally, text fields must be unflow: 
[image]

SLTS_waterlevel_1
SLTS_waterlevel_1_background

SLTS_valve_1
SLTS_valve_1_background_left,
SLTS_valve_1_background_right,
SLTS_valve_1_value
SLTS_valve_1_label

SLTS_point_1
SLTS_point_1_value
SLTS_point_1_label

SLTS_fan_1
SLTS_fan_1_background
SLTS_fan_1_value
SLTS_fan_1_label

SLTS_point_1
SLTS_point_1_value
SLTS_point_1_label

SLTS_state_1
SLTS_state_1_value
SLTS_state_1_label

sheza160217, 2023-02-17:
@Limraj :
Thanks for taking time out to reply. As I have been haviong less time to test, I have tested now and below are the results:

**_Docker Image: release-2.7.5_**
I am able to import the svg image as you had suggested. Scada-LTS creates the screen as expected. But when I click the pencil icon, the icon freezes and nothing happens.

**_Docker Image: scadalts:latest_**
The prompt to move to newer interface does not appear at all.

I have working the old interface with the "latest" docker image and so far it is good. I am working with 2 versions of docker images in parallel and if there are any major differences I will post back.

We can close this issue with this.

Thanks for your help.

closed 2023-02-17 by sheza160217

Limraj, 2023-02-17:
@sheza160217 
Does it work or are there any problems? There should be no problems with the edit button... it's worth checking.
Describe exactly what steps led to this suspension of editing. I import the corrected svg and there is no problem with editing and pinning the points. Maybe you imported a different svg? Then if the ids are ill defined then such problems may occur, this is an experimental feature.
There are no panels in the latest version 2.6.18, so you won't find these features there.

sheza160217, 2023-02-17:
@Limraj :+1: 

Ofcourse I imported the wrong .svg. I tries with the one in the above post and now it works.

Let me study the .svg file you provided. Thank you so very much!!!

Limraj, 2023-02-17:
You report a problem, so we try to help, nothing more. If you write about a bug, a scenario that will allow us to reproduce the problem is valuable to us, then we can think about what we will do with it, etc. That's all I'm after. Take care.

Limraj, 2023-02-22:
Hi @sheza160217,
If you don't use these panels, use the latest release from the tag: v2.6.18. Version 2.7.5 is not yet released and will change. 
After release/prerelease 2.7.5 version then v2.7.5 tag will be available. Each released version has a tag: v{version}; (e.g. just v2.6.18, v2.7.4.1)

Regards,
Kamil Jarmusik