SCADA-LTS/Scada-LTS #2484: Inconsistency between hierarchy added by UI and by json

fabiodurao, 2023-03-23:
Scada-LTS 2.7.4.1 - Manual instalation

1. On the hierarchy screen, I created a test folder, dragged a datapoint into this folder and it was automatically saved;
[image]
2. On the import/export screen, I checked the hierarchy checkbox, exported the json;
3. I added the XID of a variable within the json relative to the created folder;
[image]
4. Refreshing the watch list screen, the hierarchy modification was accepted;
[image]
5. Updating the hierarchy screen, the modification does not appear, even when clicking on the refesh icon available on this screen.
[image]

fabiodurao, 2023-03-23:
Update: A few minutes after making the change via json, the hierarchy screen updated, I don't understand why it took so long.
Maybe it's not the time, it's the fact that I created one more folder and updated the structure.

Limraj, 2023-03-23:
Hi @fabiodurao,
I looked at the code and it looks like the structure is updated every 10min, but the refresh button should force this update...

fabiodurao, 2023-03-24:
ok, good to know it's only a matter of time to update.

closed 2023-05-16 by Limraj

Limraj, 2023-09-11:
#2014