SCADA-LTS/Scada-LTS #2492: Map Volume Docker Application

odenisnobre, 2023-03-25:
Is it possible to map the application completely in a docker volume?
I say the files from the assets folder, uploads and etc.

Limraj, 2023-03-26:
Hi @odenisnobre,
what would you like to achieve in this way? 

You can experiment with the volumes configuration in the [docker-compose.yml](https://github.com/SCADA-LTS/Scada-LTS/blob/develop/docker-compose.yml) file:
[volumes-in-docker-compose](https://devopscell.com/docker/docker-compose/volumes/2018/01/16/volumes-in-docker-compose.html)
[volumes](https://docs.docker.com/storage/volumes/)

Regards,
Kamil Jarmusik

odenisnobre, 2023-03-26:
Actually I would like to have 'external' access to the application files.
To be able to customize some items like color, font, things like that.

The command that is in docker-compose has only mapped the logs.

odenisnobre, 2023-03-27:
I will close this case because I think I have already clarified the doubts.

closed 2023-03-27 by odenisnobre

Limraj, 2023-03-27:
Hi @odenisnobre,
You can do something like:
1. Stop scadalts container. To volumes for scadalts service add:
           - ./tomcat_webapps:/usr/local/tomcat/webapps:rw
       
[image]

2. In the place where you have placed the docker-compose.yml file, there is a create folder tomcat_webapps;
3. You download the Scada-LTS application, e.g. version beta [v2.7.5.1](https://github.com/SCADA-LTS/Scada-LTS/releases/download/v2.7.5.1/Scada-LTS.war) or stab latest [v2.6.18](https://github.com/SCADA-LTS/Scada-LTS/releases/download/v2.6.18/ScadaBR.war)
4. Transfer the downloaded file to tomcat_webapps
5. You start database and scadalts container.

This should cause the application to be deployed in folder tomcat_webapps, you will have all the contents of the application in the Scada-LTS/ScadaBR folder. If you change these files, the changes should be visible in the application.

Regards,
Kamil Jarmusik

odenisnobre, 2023-03-27:
That's exactly what I needed and I ended up getting it done by testing it over and over again.

Thank you very much.

Limraj, 2023-03-27:
@odenisnobre Just be careful with the version of the app you're using. The database structure of version 2.6.18 is different than 2.7.5.1, so if you launch the beta on the basis of version 2.6.18, you will not be able to go back to version 2.6.18.