SCADA-LTS/Scada-LTS #2502: Error characters

odenisnobre, 2023-04-05:
How to get around the character problem in an email alert message?

[image]

Limraj, 2023-04-05:
Hi @odenisnobre,
what version are you using?

Regards,
Kamil Jarmusik

odenisnobre, 2023-04-05:
Version 2.7.5.1

Limraj, 2023-04-05:
Can you provide the configuration of this alert?

odenisnobre, 2023-04-05:
I had to restart Scada-LTS and when I went back to get the settings I came across this error

[image]

odenisnobre, 2023-04-05:
I updated my docker compose here and I'm going to test it again.
the behavior is not very nice but I'll check here.
volume mapping is not working, but I'll check here

Limraj, 2023-04-05:
@odenisnobre Don't make too hasty conclusions, if you have a problem, something doesn't work, try to describe it in such detail so that someone can reproduce it, then we will be able to help.

odenisnobre, 2023-04-05:
Alright, let's go.

Firstly, the email error, this was happening until I updated the containers.
I thought updating might fix it.
But after updating volume mapping is working properly.

[image]

[image]

Limraj, 2023-04-05:
@odenisnobre, So it's working okay?

If you had a version before 2.7.4.1, this error was present, after this version, it should no longer occur, I showed you on the screenshots that it does not occur, although it is not proof that it does not occur from version 2.7.4.1, but from this version, we did not deal with this issue, and there is probably no point in checking each subsequent version, since it works in the current one. 

I also checked on version 2.7.5.1:
[image]

and result:
[image]

odenisnobre, 2023-04-05:
All good.
I believe that when emails should work.
Now talking about the docker volume, inside the webapps folders does it have to be the Scada-LTS folder, only?

Limraj, 2023-04-05:
@odenisnobre 
Theoretically it can be any name, tomcat will load the application which will be available under the name of the folder.

Running two applications on one volume means, for example, 2x the pool of connections to the database, i.e. 2x more connections. Besides, theoretically there should be no conflicts, nothing comes to mind right now. However, if it is not necessary, it is better not to keep two applications in webapps, if only because of these database connections.

odenisnobre, 2023-04-05:
OK.

So can I just leave a Scada-LTS folder inside the webapps?

Limraj, 2023-04-05:
@odenisnobre 
If the folder is properly structured and has the correct configuration files (e.g. web.xml, context.xml), tomcat will recognize it as a web application. 

`I ask because when I do the mapping, the application does not open.`
I don't understand this statement? You modify the contents of this webapps folder which is defined as bind volumes and the changes are visible in the container.

odenisnobre, 2023-04-05:
I am mapping this structure:
[image]

Is correct?

Limraj, 2023-04-05:
I don't understand what mapping means to you? It is enough that to the webapps folder, which is defined as bind volumes, copy the wara with the new version, or the folder. Then the changes should be visible from the container.

If you had the old version running before, you started the new one, the database structure is changed and you try to run the old version again, it won't work because of the changed database structure.

odenisnobre, 2023-04-05:
I'm almost at the end.
I think now the problem is in the database as the version has changed the structure as well.

But how to preserve data from the old version?

odenisnobre, 2023-04-05:
What I don't understand:

The folder structure is ok and when I access the URL:
[image]

Limraj, 2023-04-05:
@odenisnobre,
1. Which version do you want to run?
If you are unable to run this version, you can check it in:
\webapps\ScadaBR\WEB-INF\classes\version.properties.

2. I reported this in a previous issue where you asked about configuring volumes. Depends on what you want to transfer that you have to be careful. On the latest version, you can export configurations and import on a clean database on old version.
2.1. Run lastest version, and export project, generate zip file with configuration:
[image]
2.2. Next login to mysql server container:
`docker ps`
Copy container_id witch mysql server and:
`docker exec -it container_id /bin/bash`
Next backup old database: 
`mysqldump --single-transaction --routines --triggers -u root -proot scadalts > /var/lib/mysql/scadalts-$(date +%Y-%m-%d-%H).sql;`
Next drop database scadalts:
`mysql -u root -proot -e "drop database scadalts";`
And create empty scadalts:
`mysql -u root -proot -e "create database scadalts";`
Move *.sql from db_data to other path;  
2.3 Run old version application and import project:
[image]

However, problems may arise. If you don't have a strong case to go back from a newer version to an older version, then maybe there's no point in doing it.

odenisnobre, 2023-04-05:
I went back to the old version (of the folder) and the application did not load

Limraj, 2023-04-05:
@odenisnobre 
If you want us to get anywhere, try to follow my instructions. If something is unclear, ask. I asked you to check in this file what version you are trying to run. Second, the question is do you really need to go back to an older version? If you must, I have given instructions on how to transfer data from the new database structure to the old one, you must follow it. I'm just wondering if it's justified.

Limraj, 2023-04-06:
@odenisnobre I understand you have no more questions/issues, can you close the issue?

odenisnobre, 2023-04-06:
Sorry for the inconvenience.
I think I made a lot of confusion, the fact is that when I updated the stack in the portainer (docker) the application did not go up anymore
As I had made several customizations, I would go back to the previous version but, as suggested, it would not be the best option.

But no problem, I'll keep the new version but the most important thing I need is to keep the volume mapping.

I have to make my adjustments, so you can close the case, no problem.

Limraj, 2023-04-06:
@odenisnobre 
Just copy the changed files to the new version into webapps. If there is a problem, write to me, I'm closing this issue for now.

closed 2023-04-06 by Limraj