SCADA-LTS/Scada-LTS #2505: scadalts-ui dependency @min-gb/vuejs-components is not in this registry

lanicc, 2023-04-09:
**Describe the bug**
scadalts-ui dependency @min-gb/vuejs-components is not in this registry

**To Reproduce**
Steps to reproduce the behavior:

run `npm install`

```
npm ERR! code E404
npm ERR! 404 Not Found - GET https://registry.npmjs.org/@min-gb%2fvuejs-components - Not found
npm ERR! 404 
npm ERR! 404  '@min-gb/vuejs-components@0.0.7' is not in this registry.
npm ERR! 404 
npm ERR! 404 Note that you can also install from a
npm ERR! 404 tarball, folder, http url, or git url.
```

Limraj, 2023-04-11:
Hi @lanicc,
try:
1. git checkout develop, or rollback;
2. git pull;

Limraj, 2023-04-16:
@lanicc Does it work for you or can we close the topic?

lanicc, 2023-04-18:
> @lanicc Does it work for you or can we close the topic?

Thanks a lot.It's worked

closed 2023-04-18 by lanicc

stly, 2023-04-23:
@lanicc do u run this program in idea?
i deploy it but not work. 
[image]
[image]